Blockstream/Jade #116: Passphrase Option Disabled When Updating to 1.0.27

owainedwards80, 2024-02-03:
I had a Jade setup to enter a passphrase when opening. After updating to 1.0.27 the passphrase was no longer available. It goes straight to session without any option to enter passphrase. I will have to restore from seed.

JamieDriver, 2024-02-05:
I'm not sure what version you upgraded from - but that options should certainly not have disappeared!

With a jade initialised with a PIN-protected wallet, go:
`Options->BIP39 Passphrase`

You get the option to set the 'when should Jade ask for a passphrase':
Never: Don't ask for a passphrase after PIN entry
Always: Always ask for a passphrase after PIN entry
Once: Ask for passphrase if you try to unlock jade immediately/now, but in general don't (to obscure the fact that you usually use a passphrase)

NOTE: you can also set whether your passphrase is 'Manual' (ie. free text entry) or whether it is restricted to the BIP39 'Wordlist'.

JamieDriver, 2024-02-12:
Is this option ^ working for you ?  Is the issue resolved ?  Thanks.

JamieDriver, 2024-03-27:
Closing as believed to be working, and no response from raiser.

closed 2024-03-27 by JamieDriver