Blockstream/Jade #117: USB connection problem with Jade

hsdredgun, 2024-02-22:
Hi, I'm running Arch Linux and I'm having some trouble with connecting Jade to green Wallet or even sparrow via USB. 

After `sudo usermod -a -G uucp yourusername` the connection appear and everything works until  I'm trying to unlock the device then it stuck on "Checking" or the first time I tried to set it up it was then stuck on "persisting pin data". If I disconnect the usb the error message is "trouble connecting to the server". Look like Linux don't let Jade connecting to Blockstream?

This is what I tried already: 

Add your user to the dialout, plugdev, or uucp group using the following commands:

`sudo adduser yourusername dialout` or
`sudo usermod -a -G dialout yourusername`

`sudo adduser yourusername plugdev `or
`sudo usermod -a -G plugdev yourusername`

`sudo adduser yourusername uucp `or
`sudo usermod -a -G uucp yourusername`

Sparrow error message:

[image]

Thanks :)

JamieDriver, 2024-02-22:
The 'Persisting pin data' step should take a few seconds. If it's taking longer than say 10s then I'd say there's been some issue with networking comms between the app and the blind oracle, or between the app and the Jade.

If you have this repo locally, try running `./set_jade_pinserver.py` - by default it won't take any action, but it should call jade to fetch version info - this will test the arch<->jade comms (to some degree).
You could also try running `./jade_auth.py` - this will do the wallet/PIN-setup steps outside of any complex app.  If either case fails it'd be interesting to see the output.

hsdredgun, 2024-02-22:
Okay I think we got a little bit of an issue here long story short I got the exact same laptop one is a thinkpad X1 the other one is a cheap version of a thinkpad running AMD. I actually broke my X1 tonight so for science I tried with my work laptop both run Arch Linux both identical:
Install the app image of Green wallet then run the command: 
"sudo gpasswd -a uucp"
 Then 
sudo usermod -a -G uucp yourusername

And to my surprise it actually worked... So I'm even more confused... Let me try to fix my X1 and I will get back to you ASNP... Very very weird situation I'm in...

closed 2024-02-23 by hsdredgun