Blockstream/Jade #119: Show Liquid address in readable format

Donach, 2024-02-29:
When using Jade + Green (but applies to any Liquid wallet I guess), when validating the address on Jade, the format is very unreadable - just really long text smashed together, and also separated to two screens by just mere few characters.

I would propose to display the string in more readable way in 4-5 char chunks, as follows (using some latest confidential TX from the explorer):
ex1p0 u7uzf 9mv9p 
gerr6 yys7r ytdas 
w2gpm ks8lc 8tws0 
qgc9t dsznc qsht7fu 
(last chunk would be longer, if it cannot fit into two full 4 or 5 char chunks).

This would improve readability and reduce errors when validating - also, same formatting would be nice if reflected in Green wallet, so user can check individual "chunks" as they would be on the same position, it would be easy to find and match the ones from Jade in Green.

JamieDriver, 2024-02-29:
I believe there is a change in the works to display all addresses in this format - thanks for the upvote!  ;-)

Donach, 2024-02-29:
Sorry if this is duplicate issue then - I did not find it among open ones...
If it's not, then I'll be happy to check the progress here :-)