Blockstream/Jade #121: Git Tags for newer versions

1-21gigasats, 2024-03-24:
Hi, i am planning to give a workshop for building jade-diys based on the jade diy installer (https://github.com/epiccurious/jade-diy). The Installer uses git tags to build the latest version. The latest available tag is 1.0.27 but according to your changelog the latest released version is 1.0.29. Could you create git tags for this version so we can build on the newest available version or isn't this version meant to be used on production devices?

JamieDriver, 2024-03-25:
Hi,
1.0.29 is just in the final stages of internal testing (it's only a small bugfix for liquid on top of 1.0.28) - I will push the tag as soon as it is agreed/finalised - should only be a a day or two, tops.
I have checked all other tags present.  Many thanks.

1-21gigasats, 2024-03-25:
Perfect, thank you

JamieDriver, 2024-03-27:
fyi. `1.0.29` formally released and the tag pushed.  :+1:

closed 2024-03-27 by JamieDriver