Blockstream/Jade #123: As a new user trying to connect Jade to Green, I cannot connect to the blind oracle because the domain redirect to j8d.io displays as a Warning

aassoiants, 2024-03-30:
Friend is experiencing issues connecting their Jade to their Android Green. 

See images for context
[image]
[image]

Reference code: https://github.com/Blockstream/Jade/commit/05d87361ffc90ab3d0097255cc73784e3720bfa1

JamieDriver, 2024-04-02:
Rest assured `j8d.io` is a valid pinserver url.

This warning should be fixed in Green desktop app `v2.0.3`

`j8d.io` is a synonym for `jadepin.blockstream.com`, preferred in recent fw releases - so using a recent jade fw with a 'not-latest' app may show this warning.

```
$ nslookup j8d.io
...
Name:	j8d.io
Address: 35.237.55.41

$ nslookup jadepin.blockstream.com
...
Name:	jadepin.blockstream.com
Address: 35.237.55.41
```
You can either upgrade to desktop app 2.0.3, or if preferred can safely accept this oracle url.

aassoiants, 2024-04-02:
This was on the latest Android app

Sounds like the team might need to do a new release

bitcoinhelp, 2024-04-02:
> This was on the latest Android app
> 
> Sounds like the team might need to do a new release

Which version are you on? The error will not appear in 4.0.27

bitcoinhelp, 2024-04-02:
> Friend said they were running the latest F Droid version

F-droid has not yet been updated to my knowledge so that makes sense. They can choose to allow the non-default connection in the meantime to use Jade normally

closed 2024-09-12 by JamieDriver