Blockstream/Jade #127: OTP - add support for google authenticator exported QRcode ?

Drew72-ita, 2024-04-01:
This report/suggestion concerns the acquisition and decoding of the QR code containing the seed/secret for OTP.

I see that the standard format (encoded in the QR code) is recognised:
otpauth://totp/Example:alice@example.com?secret=JBSWY3DPEHPK3PXP&issuer=Example
(this is an example, there is no sensitive data)
which you can also enter by hand as a text string (I tried it and it works)

However, it does not recognise the fairly widespread format used to export OTPs by applications such as google authenticator, which encodes a string in the QR code in the format
otpauth-migration://offline?data=CiMKCskCjMDVZlPV3OQSD015IFRlc3QgQWNjb3VudCABKAEwAhAB
(this is an example, there is no sensitive data)

The use case is an user exporting the OTP from "hot" and relatively unsafe  google authenticator to a safer cold HW wallet as jade.

JamieDriver, 2024-04-02:
This is a great suggestion!
I'll see if I can find documentation relating to this format, and if so (ie. I can parse what's in the 'data' element) will look to add support.
Many Thanks.

Drew72-ita, 2024-04-02:
Hi,

I'm glad to be useful ^_^.

I used this

https://github.com/dim13/otpauth

but for command line conversion.

JamieDriver, 2024-04-02:
Right, it appears to use protocol-buffers, which may not be something we can support immediately.  More digging will be required.