Blockstream/Jade #128: Jade issues with Specter signing

Dyor69, 2024-04-03:
When I try to send a transaction with Specter Desktop I get get a message that ' device does not have matching signing key'
Also wont let me verfiy addresses
Not sure where the issue is  as I have sent transaction without any issues in the past. 
Any advice is greatly appreciated .

JamieDriver, 2024-04-03:
That error must be coming from Specter - not one of ours.  
So difficult to diagnose at this point as not sure how that error message arises.  
Is it related to the version of the jade fw you are using, the version of the Specter install, or is it specific to a particular transaction ?
Any additional info would be very useful - thanks.

JamieDriver, 2024-04-03:
From what I can gather, it looks like the seed Specter is expecting to see on the hw wallet is not the seed that is currently on there.
eg.  Specter is expecting the root xpub/fingerprint to be <xxx>, but when it requests that from the hw, it gets <yyy>

( fyi: when unlocked, Jade should display the wallet fingerprint in hex in the bottom-right of the screen.  The fingerprint is a short id for the mnemonic phrase / seed of the wallet. )

Dyor69, 2024-04-04:
Thanks for you response Jamie.
Looks like its a fingerprint issue
Here's the message I get to verifiy receiving address
[image]
And the send one
[image]
This is the version of Specter
Version: v2.0.2rc8.dev0+g23ad1197.d20230921     UID: 1bf3305caf91e16b
JAde Firmware
1.0.29
Would you like me to paste the Jade xpub key here?

Thanks
Campbell

JamieDriver, 2024-04-04:
I don't think there's anything I can do - this isn't an issue with Jade firmware - unless you're telling me (for example) it works with v1.0.27 but doesn't work with v1.0.29.  
It *looks like* the Specter wallet was set up (with Jade?) with a particular seed/mnemonic, and now that Jade has a different seed/mnemonic (or is a different Jade?).

These messages appear to be saying, 'you've plugged the wrong hw unit' ?

Dyor69, 2024-04-04:
Thanks Jamie
It’s very strange as this is the only Jade I own and nothing has changed
since I set it up and did test transactions before adding funds to the
wallet.
Is there a way to check I have the correct pass phrase. It’s the only thing
I can think of. Otherwise it’s a Specter issue. Thanks again for responding

JamieDriver, 2024-04-04:
I assume you know what your 'recovery phrase words' are.  [**DO NOT post them here (or anywhere!) ].
The 'fingerprint' you see in the bottom right is a short 'summary' (if you like) of that phrase (plus any bip39 passphrase).  ie. a short id of the wallet.  [ If you are using a bip39 passphrase you must be very careful to enter the exact same phrase each time. ]
Jot down that fingerprint (8 hex chars).

With Jade in 'unlocked' state, go 'Options->Temporary Signer' and enter the 'recovery phrase' words and any bip39 passphrase.
Connect to Specter and the Jade should unlock - it will show you the 'fingerprint' of the wallet you have just entered.
It should be the same as the one you recorded earlier above.

Yes, if the bip39 passphrase was entered even slightly differently, it will lead you to a different wallet, different fingerprint/xpubs, etc.  If you've used this passphrase, I guess you could try slightly different variations of phrase, or spelling, capitalisation, etc. ...

Unfortunately with this stuff there is no way to go backwads eg. from fingerprint or xpub to recovery phrase or bip39 passphrase etc.

Dyor69, 2024-04-06:
Thanks Jamie
I new it was my error, so thanks for helping me narrow it down. 
All sorted out now, it was the slightest difference in the passphrase. So sorry to waster your time.
It goes to show how careful you have to be to record everything accurately.

JamieDriver, 2024-04-08:
Glad it all worked out for you!  (and doubly glad it wasn't a bug in Jade fw! :rofl:)  
Nice one :+1:

closed 2024-04-08 by JamieDriver