Blockstream/Jade #129: Wen codex32?

seedhammer, 2024-04-04:
We're planning a product for generating seeds by [manual entropy](https://github.com/BlockstreamResearch/codex32/issues/67), and for that purpose the small 32 character alphabet of codex32 is a much better fit than the 2048 alphabet of BIP39, on top of the other advantages of codex32. 

To kick off codex32 adoption, the Jade seems like the ideal trailblazer: both it and codex32 are from Blockstream. So, as the issue title asks, is codex32 planned for Jade? If not, is there anything we can do to help?

BenWestgate, 2024-09-24:
I'm writing an example implementation for electronic codex32 share set generation for my project github.com/benwestgate/bails that will help. I also have written a CodexQR specification that allows encoding codex32 shares in 21x21 QR codes like CompactSeedQR.

I'm not sure why Jade doesn't have a codex32 PR yet, we finished the [codex32 import document](https://github.com/BlockstreamResearch/codex32/blob/master/docs/wallets.md) a year ago. They probably want both creation and import specified before doing it.

BenWestgate, 2026-10-02:
I'd like to turn this into a concrete funded Jade milestone.

I've already implemented codex32 in my own projects and contributed to the codex32 wallet integration documentation and BIP work.

**Proposed milestone: codex32 restore support**

Implement codex32 wallet recovery in Jade's Restore Wallet flow according to the behavior specified in `BlockstreamResearch/codex32/docs/wallets.md`, including validation, threshold recovery, error correction, and insertion/deletion correction.

Deliverables would include the Jade firmware/UI integration, tests, and documentation needed for production review.

Backup generation, resharing, and QR share formats can remain separate later milestones.

I'm available to implement this as a paid fixed-price milestone or contractor engagement. If Jade/Research wants to proceed, could one of the maintainers point me to whoever can approve the scope and budget?

Once funding is agreed, I'll deliver the work as reviewable PRs.

jgriffiths, 2026-10-02:
@bitcoinhelp Do you mind taking a look at the scope and suitability of this feature?