Blockstream/Jade #131: Untrusted Hash found

eliaspfeffer, 2024-04-14:
Couldn't find the following hash on this list:

_ "Update der Firmware...
Hash: **fd4e3fc1 566c9016 f0218ad9
5bca6994 39d0b509 99fd5aa6
ae848a97 699f73c3**"_
Version: 1.0.29, updated via Bluetooth.

I've tried to find it here:
https://jadefw.blockstream.com/bin/jade/index.json

Is my hash trustworthy? Do I have to check the hash maybe somewhere else?

How to reproduce:
Brand new Jade, updated via iOS AppStore Green App, this Friday (12th of April 2024).

JamieDriver, 2024-04-15:
If your Jade has a 'rocker', rather than a 'true wheel', it is a Jade revision 1.1.  (Only a relatively small number of 'true wheel' units were made).
So you should probably look in: https://jadefw.blockstream.com/bin/jade1.1/index.json

Specifically, here - full fiel upload (ie. not a delta/patch), updating to 1.0.29 fw, variant including Bluetooth drivers.
```
  "stable": {
    "full": [
      {
        "filename": "1.0.29_noradio_1118208_fw.bin",
        "version": "1.0.29",
        "config": "noradio",
        "fwsize": 1118208,
        "cmphash": "b0b06a3b280361d4a00d7037579c30ff0935e97595e22155ca7e6823a7341a2c",
        "fwhash": "aee4feff3ea1ee591ac1ab3f283ac828fcedd38d53f69676254cd23c206ef9be"
      },
      {
        "filename": "1.0.29_ble_1445888_fw.bin",
        "version": "1.0.29",
        "config": "ble",
        "fwsize": 1445888,
        "cmphash": "fd4e3fc1566c9016f0218ad95bca699439d0b50999fd5aa6ae848a97699f73c3",
        "fwhash": "673054922e0e2b6e31680fa1317c4245ae5b941f5c552564798457e455dca62f"
      }
    ],
```

JamieDriver, 2024-04-15:
I don't see an issue here - pls close if you agree.  Many thanks.  J.