Blockstream/Jade #134: Feature request: add display option to flip vertically the screen

Drew72-ita, 2024-05-13:
Request to add settings -> display option to flip vertically the screen:
- facilitate the use  buttons for left-handed users
- for DIY builds the usb port / buttons could be on the "wrong" side for optimal use
- actually a common option for ESP32 firmwares (see LoRa,Meshtastic etc)

JamieDriver, 2024-05-13:
I assume you mean 'rotate 180 degrees' ?
If so, we partially did this from the outset, but at the time veered away from it due to concerns about the display not being quite right (and ofc having 100 other more pressing things to implement!)
Eagle-eyes can still spot code related to screen orientation and inverting wheel/buttons in the codebase, so as you suggest it should be possible to reanimate that.
I'll take a look and we'll discuss priorities internally.

Drew72-ita, 2024-05-13:
Yes, sorry 180 deg. rotation is more correct.
ty ^_^

JamieDriver, 2024-05-28:
This should make it into the next release, or worse case the one after :+1:

Drew72-ita, 2024-05-28:
Wow thank you very much ^_^

JamieDriver, 2024-08-15:
Added to fw `1.0.31` for diy hardware.

> If so, we partially did this from the outset, but at the time veered away from it due to concerns about the display not being quite right (and ofc having 100 other more pressing things to implement!)

Yes, there is an issue with the screen used for older Jade hw which has causes artifacts if we flip the screen orientation.
So atm only enabled for 'diy' hardware.

Future batches of proper Jade hw have a better screen without this issue - this will be detected and 'flip screen' will be enabled in this case...

Apologies it's not 100%...

Drew72-ita, 2024-08-15:
Thank you, I will test it on my next firmware update : )

closed 2024-09-12 by JamieDriver

Drew72-ita, 2024-09-24:
Tested and seems fine.

Just out of curiosity: was this new feature the reason the JADE splash logo was removed at startup?
(maybe it interfers with images loading at startup?)

JamieDriver, 2024-09-24:
No, unrelated :-)
Maybe partly because image data is now kept zipped in the binary, and has to be unzipped in RAM, which some 'diy' devices don't have much of ... and maybe partly to differentiate official jade devices from 'diy' ... not totally sure tbh.