Blockstream/Jade #136: Can not update

jurgen2005, 2024-05-17:
When I run " python jade_ota.py --noagent " 
The TTGO asks for a pin, then shows the old and new firmware to confirm
then very shortly shows the progress bar and then shuts off..

any idea what this could be?

JamieDriver, 2024-05-17:
Hi - Where did the fw image you are trying to flash come from ?

jurgen2005, 2024-05-17:
Hi, its a ttgo-tdisplay running 1.0.30-4-g29
I'm just playing with it an allready updated it several times. It has the secure bootloader as described here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html
I flash it on Ubuntu with the IDF tools.
Today I did a git pull and now it wont accept the new firmware, I have two units and the both show the same behaviour.

JamieDriver, 2024-05-17:
OK, so once you've enabled 'secure-boot' you need to sign every firmware image - presumably you have a personal key you are signing with ?
Ok, so you're running commit '2957b92b0620fe53909fbac00a3d1d5ab886a92c' and are trying to update to a later master commit ?
I assume you build the fw locally, then sign it with the correct key, and are then trying to OTA it ?

jurgen2005, 2024-05-17:
Yes, the signing key is in the sdkconfig.defaults. 
Building log shows 1 signing key found

jurgen2005, 2024-05-17:
pastebin.com/qDtidaDx

JamieDriver, 2024-05-17:
do you also use flash-encryption ?

JamieDriver, 2024-05-17:
I did an ota on my tdisplay with (without secure-boot/flash-encryption - default tdisplay config) from the version you suggest to latest master and it worked ok:
```
jamie@athens:~/blockstream/jade$ ./jade_ota.py --skipble --fwfile-uncompressed ./build/jade.bin
Reading file ./build/jade.bin
Read 1418048 bytes
Compressing 1418048 bytes
Compressed to 910520 bytes
Deduced compressed firmware filepath: build/1.0.30-10-gbb0cbb9c_ble_1418048_fw.bin
Filename suggests full firmware : 1.0.30-10-gbb0cbb9c_ble_1418048_fw.bin
Got fw file of length 910520 with expected uncompressed final fw length 1418048
Final fw hash: 0ec4d6800f986933888a971d8f326e8c206d67b9caca07a90e61b310c728c979
Jade OTA over serial
Sending: {'method': 'get_version_info', 'id': '213834'} as cbor of size 35
Received msg: {'id': '213834', 'result': {'JADE_VERSION': '1.0.30-4-g2957b92b-dirty', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'TTGO_TDISPLAY', 'JADE_FEATURES': 'DEV', 'IDF_VERSION': 'v5.1.4', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': '3C61050C6474', 'BATTERY_STATUS': 0, 'JADE_STATE': 'UNINIT', 'JADE_NETWORKS': 'ALL', 'JADE_HAS_PIN': False, 'JADE_NVS_ENTRIES_USED': 21, 'JADE_NVS_ENTRIES_FREE': 357, 'JADE_FREE_HEAP': 91376, 'JADE_FREE_DRAM': 91376, 'JADE_LARGEST_DRAM': 90112, 'JADE_FREE_SPIRAM': 0, 'JADE_LARGEST_SPIRAM': 0}}
Running OTA on: {'JADE_VERSION': '1.0.30-4-g2957b92b-dirty', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'TTGO_TDISPLAY', 'JADE_FEATURES': 'DEV', 'IDF_VERSION': 'v5.1.4', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': '3C61050C6474', 'BATTERY_STATUS': 0, 'JADE_STATE': 'UNINIT', 'JADE_NETWORKS': 'ALL', 'JADE_HAS_PIN': False, 'JADE_NVS_ENTRIES_USED': 21, 'JADE_NVS_ENTRIES_FREE': 357, 'JADE_FREE_HEAP': 91376, 'JADE_FREE_DRAM': 91376, 'JADE_LARGEST_DRAM': 90112, 'JADE_FREE_SPIRAM': 0, 'JADE_LARGEST_SPIRAM': 0}
Sending: {'method': 'ota', 'id': '770118', 'params': {'fwsize': 1418048, 'cmpsize': 910520, 'cmphash': '8a1dd487d3de2ff119e029dc1c9969b122e5c6cbd21955ae39d5683b61c7d9f7', 'fwhash': '0ec4d6800f986933888a971d8f326e8c206d67b9caca07a90e61b310c728c979'}} as cbor of size 138
...
Received msg: {'id': '569334', 'result': True}
Total ota time in secs: 138.88631510734558
```

So issue probably related to 'secure' features ...

JamieDriver, 2024-05-17:
I assume you've tried a few 'destination' versions ?  ie. you can't ota back to 1.0.30 ?  
Do you recall what version you flashed originally, and which ota's worked ... ?

jurgen2005, 2024-05-17:
I will try to go back! I came from 1.29. What is flash-encryption? I'll show the config I use

jurgen2005, 2024-05-17:
https://pastebin.com/L7yun3mk

jurgen2005, 2024-05-17:
1.29 needs a lower idf 5.1.2

jurgen2005, 2024-05-17:
no matter what machine I connect to the TTGO it goes dark, like with this tool: [https://espressif.github.io/esptool-js/](https://t.co/axsbS10HyI)

nate922, 2024-06-06:
@jurgen2005 I think I'm getting the same issue with the m5stickC, after confirming it starts, then reboot -
 
No further logging on the console (--log debug) - I'm also using secureboot. Did you find a solution with this one? Cheers

jurgen2005, 2024-06-07:
Nope no solution, for now they are stuck in this version. I think the problem is in the device firmware itself, so might be impossible to fix...

nate922, 2024-06-08:
I tried using the tool you mentioned ttps://espressif.github.io/esptool-js/ and it bricked, lol

Eventhough I have the PEM file I don't think it's recoverable as the key is on the unit itself.

grezzogg, 2024-06-24:
Hello.

This happened to me too when I installed 1.0.30-18-g7 on my lilygo. The device works, it just won't update anymore, it just reboots. Found any solution @jurgen2005 ?

I have another lilygo running 1.0.30 and not I'm scared of updating again. Should I just wait for 1.0.31?

jurgen2005, 2024-06-24:
No, no solution, think it will be stuck on this version forever :-/

grezzogg, 2024-06-24:
I also have a second unit but It's on version 1.0.30. Should I wait a few releases to go by before updating? What happens if I don't update at all?

jurgen2005, 2024-08-25:
If we don't update at all, everything will still work I thing. Up to a certain point.

JamieDriver, 2024-09-03:
#151 resolves a (different) issue doing this with an M5StickCPlus2 device, so we know this secure+ota pattern works with 'production' jade units, and with the StickC ... the issue *could be* related to the tdisplay hw ... is that the hw you are all experiencing issues with ?