Blockstream/Jade #138: Firmware Extraction Attack

legioz, 2024-06-08:
I recently came across [this article](https://www.ledger.com/blog/firmware-extraction-evil-maid-attacks-on-blockstream-jade-hardware-wallet) that explains how it is possible to extract the seed phrase encrypted in Jade if the attacker manages to steal the device.
Is there already a way to deal with this problem? I looked through the commits here, but I didn't see any solution so far

3rdIteration, 2024-06-17:
This doesn't extract the seed, just the allows the decryption of the credentials used with the blind pinserver. (So someone could do something that would wipe your pin there and need you to reinitialize the device from your recovery seed)

legioz, 2024-06-17:
@3rdIteration isn't the encrypted Private Key also stored locally in the hardware wallet?
If one can decrypt the PIN, it makes sense that he would be able to decrypt the private key saved on the same device, doesn't it?

3rdIteration, 2024-06-17:
No, not with Jade. It needs a decryption key from the blind PINServer. (Which needs the PIN)

This is why it never mentioned that it could extract seeds or private keys.

closed 2024-06-17 by legioz