Blockstream/Jade #142: Improve UX for descriptor registration

pythcoiner, 2024-06-24:
While testing the Jade implementation in Liana wallet i notice 2 important points:
In miniscript descriptors it's forbiden to reuse a key w/ the same multipath, so in liana we change the multipath for each occurence (<0;1>, <2;3>, <4;5>, etc...) when a key is used in several conditions. Multipaths are displayed in the descriptor template while registering a descriptor on jade, that is what i expected:

[image]

but multipath are also displayed while reviewing the keys:

[image]

- it is unnecessary and add a lot more in what the user need to verify, in the picture example, each key(total 4 keys) is use 3 time in the descriptor, so i have to verify 12 keys where in fact there is only 4!

- the jade never notify the user which key it owns, but it should!

JamieDriver, 2024-06-24:
1. Point taken, yes am aware it's not ideal, will try to look into it asap

2. Yes, should definitely highlight that in some way, agree.  I think in 'multisig' it just sticks a asterisk in the title bar - will see if we can do something similar. 

Many thanks.

pythcoiner, 2024-06-24:
> 2\. Yes, should definitely highlight that in some way, agree.  I think in 'multisig' it just sticks a asterisk in the title bar - will see if we can do something similar.

i think you can replace the path by wether the key is owned or not

pythcoiner, 2024-06-25:
> Just to check though, do you agree you do see this '*' for your signers ?

yes got it!