Blockstream/Jade #143: Do not display change output as a payment

pythcoiner, 2024-06-24:
When signing a transaction, the change ouput should be at least labelled as a change (or not displayed at all), w/ the current implementation both ouputs (external & change) are displyed in the same way, so the user cannot be sure if the change go to itself or someone else.

JamieDriver, 2024-06-24:
My understanding was that change outputs were internally validated and not shown to the user at all (for singlesig and for registered multisigs and descriptors).  
There certainly *appears to be* code trying to handle that case.  I'll try to look into it.

pythcoiner, 2024-06-24:
> My understanding was that change outputs were internally validated and not shown to the user at all (for singlesig and for registered multisigs and descriptors).

i think its better yes, in order to reduce amount of infornation to be validated by the user

JamieDriver, 2024-06-24:
I assume you're registering a descriptor, then using 'sign_psbt' ? [ just so I can try the same thing here ... ]

pythcoiner, 2024-06-24:
> I assume you're registering a descriptor, then using 'sign_psbt' ? [ just so I can try the same thing here ... ]

yes

JamieDriver, 2024-06-25:
Ok, I'm not seeing the change output, as I'd expect...
I register a (pretty simple 2of2 multisig) descriptor, then sign a psbt which has a change output.  On Jade hw I only need to confirm the spend output (and the fee).  I don't see the change, which Jade has verified internally (by rebuilding the script/address from the descriptor).

Possible causes:
- the descriptor is tied to the signer - ie. jade must be unlocked with signer X when the descriptor is registered, for that descriptor to be used to verify change when signer X is signing the psbt.    ie. if you are switching signers on your jade hw a lot, you may have registered the descriptor with X but are then signing with Y, and wondering why the change isn't being detected ?    ( even if X and Y are both party to the policy )
- the descriptor is tied to the network - ie. if you register a descriptor for 'testnet' it can't be used on 'mainnet' - although ofc if using a PIN-protected wallet that's also tied to the network.  But if using a 'temporary signer' you may have registered the policy against 'mainnet', and later tried to use it on 'testnet' ?
...
I can't off the top of my head think of any other reason for the record not to be used (and/or the change address not be internally verified).
ofc the slightest difference in the descriptor registered vs the one used to construct the psbt ?

pythcoiner, 2024-06-25:
i'm testing w/ the most basic descriptor liana can produce:

wsh(or(pk(a), and(pk(b), older(timelock))))

i can prepare some test cases if you need or reproductible steps w/ [async-hwi](https://github.com/wizardsardine/async-hwi) cli

JamieDriver, 2024-06-25:
This works for me - it uses the seed phrase public in the jade tests - you'll need to load that into the jade with a 'testnet' network.
eg. ensure the jade is currently uninitialised/has no PIN-saved wallet, run the script and then setup jade using 'restore' and 'scan qr', and paste the words into eg. https://www.qrcode-monkey.com, and scan the qr it provides.  As I say, this isn't a 'sensitive' mnemonic phrase.

[jade_test_psbt_change.py.txt](https://github.com/user-attachments/files/15970336/jade_test_psbt_change.py.txt)

I am hoping this will then work for you!

If you can adjust it to not work when you think it should, eg. with your descriptor and psbt (but still using this wallet phrase so I can run exactly the same thing!) I'll take a look asap.

pythcoiner, 2024-06-25:
Not yet test your way but i just do this:
- factory reset
- create a key w/ jade
- create this wallet w/ liana(signet):
```

```
- register the descriptor
- fund it
- make this spend w/ change => only external output is displayed:
```
cHNidP8BAIACAAAAAYDkmCQHzxAK/DnBgfPki8T+Em6j+ZtLi+soR9c24L2BAQAAAAD9////AiChBwAAAAAAGXapFLpBIIOhrYHBuSCTAa87QcKbPIEZiKxzoAcAAAAAACIAIMNfT4T+ZOvHXQvVSEZaQh+pmcBRGJqeoo+AWivHUhRiAAAAAAABAP19AQIAAAACSyZJgUr8MPxJj2zZOCbxa7fKqjiFp3bTBkjYybNfke4AAAAAakcwRAIgZI3QqrlWMSC9A8h5bwIyaJo1AgvyDqv+xSbcpS+muXsCIC/i9eA4/qbdJ/oYrIH6Is3A65BXJNnEZHh2/DyrEznuASECvtQ15k8KAGwJuertrrq1wm+3YWdFwRcUf3F+rK7XQuj9////RMvdZebzlD+TdzqKVwRnSKl2EuJfFBUQZEJSYQAlhEwAAAAAakcwRAIgYYo1qVEvnmAfep+/zS7kIKyTX2cMR8d/ml4mErmQth4CIC5oiQdO04MHie34lZKliWXJ8BTELHNcYpc+8TZ4C7faASECdE4CTymAj9/LjA8hQuf/6zyxxoyduofF7/fxBBaLHyX9////AmgFDgAAAAAAGXapFLOCdinP84R3Ji4KAzUo+RGQBWD/iKxAQg8AAAAAACIAIJAOpQfS/4/3xnimkOxnURfOXCSg7LW1ipM8ODRxptyVSRMDAAEBK0BCDwAAAAAAIgAgkA6lB9L/j/fGeKaQ7GdRF85cJKDstbWKkzw4NHGm3JUBBUQhAlDeAs6Gm9Wn0NpqyJoUQoBYuvgT4IOaZuBVf2Co/e4PrHNkdqkU/RSnc056LkhL0vQs81ZxFfkmVw2IrQP//wCyaCIGAlDeAs6Gm9Wn0NpqyJoUQoBYuvgT4IOaZuBVf2Co/e4PHLq9R+QwAACAAQAAgAAAAIACAACAAAAAAAAAAAAiBgM8AXD63Nsj0r6JEmfQL9QwRxAnd6CwFaiDbJoiMpoL+Bw33sCAMAAAgAEAAIAAAACAAgAAgAAAAAAAAAAAAAAiAgK6hS9sKWIfmR9AL46dEHStzoo58YKmlha5j47YSiXvbRw33sCAMAAAgAEAAIAAAACAAgAAgAEAAAABAAAAIgIDV2Lm09DmljfNvDtUM50/sulQHF0vlv159XyEjFTIPHAcur1H5DAAAIABAACAAAAAgAIAAIABAAAAAQAAAAA=
```
- make this spend (send to self) => Output to self is displayed (should not)
```
cHNidP8BAF4CAAAAAYDkmCQHzxAK/DnBgfPki8T+Em6j+ZtLi+soR9c24L2BAQAAAAD9////AbVBDwAAAAAAIgAgdXvOKcv14M2HrF01e+sslLH2SL0yEQecavWVS2AzP5IAAAAAAAEA/X0BAgAAAAJLJkmBSvww/EmPbNk4JvFrt8qqOIWndtMGSNjJs1+R7gAAAABqRzBEAiBkjdCquVYxIL0DyHlvAjJomjUCC/IOq/7FJtylL6a5ewIgL+L14Dj+pt0n+hisgfoizcDrkFck2cRkeHb8PKsTOe4BIQK+1DXmTwoAbAm56u2uurXCb7dhZ0XBFxR/cX6srtdC6P3///9Ey91l5vOUP5N3OopXBGdIqXYS4l8UFRBkQlJhACWETAAAAABqRzBEAiBhijWpUS+eYB96n7/NLuQgrJNfZwxHx3+aXiYSuZC2HgIgLmiJB07TgweJ7fiVkqWJZcnwFMQsc1xilz7xNngLt9oBIQJ0TgJPKYCP38uMDyFC5//rPLHGjJ26h8Xv9/EEFosfJf3///8CaAUOAAAAAAAZdqkUs4J2Kc/zhHcmLgoDNSj5EZAFYP+IrEBCDwAAAAAAIgAgkA6lB9L/j/fGeKaQ7GdRF85cJKDstbWKkzw4NHGm3JVJEwMAAQErQEIPAAAAAAAiACCQDqUH0v+P98Z4ppDsZ1EXzlwkoOy1tYqTPDg0cabclQEFRCECUN4Czoab1afQ2mrImhRCgFi6+BPgg5pm4FV/YKj97g+sc2R2qRT9FKdzTnouSEvS9CzzVnEV+SZXDYitA///ALJoIgYCUN4Czoab1afQ2mrImhRCgFi6+BPgg5pm4FV/YKj97g8cur1H5DAAAIABAACAAAAAgAIAAIAAAAAAAAAAACIGAzwBcPrc2yPSvokSZ9Av1DBHECd3oLAVqINsmiIymgv4HDfewIAwAACAAQAAgAAAAIACAACAAAAAAAAAAAAAIgICC9xFcahlGNMHygdbx40QjKDAbB8/c51r9wpFQXt7Pjocur1H5DAAAIABAACAAAAAgAIAAIABAAAAAgAAACICA2H57YmgH4bugaKgaO4DtXYcwTUA3O7ZOs6ciT6BDiz0HDfewIAwAACAAQAAgAAAAIACAACAAQAAAAIAAAAA
```

will dig more this tmr, maybe i'm hitting an edge case

JamieDriver, 2024-06-25:
btw I think that's the quick acid test of whether it should work ... look at the 'cli decode' of the psbt, and see the address of the change output in the tx details (at the top of the) - eg. for my case above:
>
    "vout": [
      {
        "value": 0.00000842,
        "n": 0,
        "scriptPubKey": {
          "asm": "0 209de60270426474752d1553da556442e3cd8e9ba3f0b6787f341caaf3a18e2c",
          "hex": "0020209de60270426474752d1553da556442e3cd8e9ba3f0b6787f341caaf3a18e2c",
          "address": "bcrt1qyzw7vqnsgfj8gafdz4fa54tygt3umr5m50ctv7rlxsw24uap3ckqd3g4py",
          "type": "witness_v0_scripthash"
        }
      },
      {
        "value": 0.00001000,
        "n": 1,
        "scriptPubKey": {
          "asm": "OP_HASH160 b7c56de7fde5a631cbb4fc284587c09d9efc6c4f OP_EQUAL",
          "hex": "a914b7c56de7fde5a631cbb4fc284587c09d9efc6c4f87",
          "address": "2N9zvA22xVyCxivNJ1BfY4pf1tJuxHHmBMu",
          "type": "scripthash"
        }
      }
    ]
  },
<
The change is the `vout: 0` 842 sats output - `bcrt1qyzw7vqnsgfj8gafdz4fa54tygt3umr5m50ctv7rlxsw24uap3ckqd3g4py`.

From the `outputs` section (at the bottom), we can get the key paths:
```
"outputs": [
    {
      "witness_script": {
        "asm": "2 02c99198821d69e326df0399c7698b8cd20fa25f32cdcc799bf076ec4fbdc0810c 032332c651a545260b82b1c9f2dbd65b4c15a73ba61ada165ad468e4411ada4538 2 OP_CHECKMULTISIG",
        "hex": "522102c99198821d69e326df0399c7698b8cd20fa25f32cdcc799bf076ec4fbdc0810c21032332c651a545260b82b1c9f2dbd65b4c15a73ba61ada165ad468e4411ada453852ae",
        "type": "multisig"
      },
      "bip32_derivs": [
        {
          "pubkey": "02c99198821d69e326df0399c7698b8cd20fa25f32cdcc799bf076ec4fbdc0810c",
          "master_fingerprint": "e3ebcc79",
          "path": "m/48'/1'/0'/2'/1/8"
        },
        {
          "pubkey": "032332c651a545260b82b1c9f2dbd65b4c15a73ba61ada165ad468e4411ada4538",
          "master_fingerprint": "0a5b42f6",
          "path": "m/48'/1'/0'/2'/1/8"
        }
      ]
    },
    {
    }
  ],
  "fee": 0.00000700
```

We can then see whether jade+descriptor will generate that address, using that path suffix:
```
    addr = jade.get_receive_address('localtest', 1, 8, descriptor_name=DESC_NAME)
    assert addr == 'bcrt1qyzw7vqnsgfj8gafdz4fa54tygt3umr5m50ctv7rlxsw24uap3ckqd3g4py'
```

If addresses not same, then there's either something wrong with the descriptor registration, or something wrong with the psbt - in that they are not consistent.

If they are the same, but Jade still isn't auto-verifying the change, there's def something for me to look into.

JamieDriver, 2024-06-25:
Ahh, it's a send-to-self / redeposit ?  That may be confusing matters as it might be thinking it can automatically verify both outputs, leaving nothing to show the user at all - which it will refuse to do and fail-over to showing them everything.  
I'm not 100% but will dig further / try to run your case above, but I seem to recall there may be some issues in this area.

[ The 'fail safe' default behaviour with Jade is that if there is any uncertainty it shows the output to the user to confirm. ]

pythcoiner, 2024-06-26:
> In this case while Jade could verify and hide that output, it still shows the output to the user to agree - otherwise there's nothing to show at all and the user wouldn't know what they were signing. ;-) - just that the fee was 139 sats.

I think you can display it if you notify user it'a a send-to-self.
As user can only trust the signing device screen, he does not have a way do be sure he send money to a owned address if you don't display it...

pythcoiner, 2024-06-26:
> otherwise there's nothing to show at all and the user wouldn't know what they were signing.

The way ledger do is not bad i think, they just notify you you'll spend and show you the wallet name

JamieDriver, 2024-06-26:
Right, so hiding normal 'change' output on an external spend is fine ... but when it's a re-org/consolidation and we show the outputs to self to the user to confirm, highlight it in some way that it's clear it's the wallet's own address.

There is a 'banner' area under the output details, should be easy enough to poke a message in there.

JamieDriver, 2024-06-26:
Jade detects 'own wallet address' by trying to re-create the script/address of the output from the info given in the descriptor registration plus the key origins/paths given in the psbt.

pythcoiner, 2024-07-01:
Here steps to reproduce what @edouardparis describing:

- [x] Jade 
  - bulb moment cry reform bitter virus shadow custom hockey kid delay quantum
  - [8bad162e/48'/1'/0'/2']tpubDDycBkTWNrqFprvKr2mFUJYtSDYHckrhXqv6ckMZKzhTZNgYoTMHaX5oYitKdxXTiYK1BePhKScMYFqR6qR4G3MNnyD35N9xK8kZ7Bo6xxP

- [x] Liana Hot key 
  - impact outer oven early cost radar spend culture stand salon oval badge
  - [20aa8ca2/48'/1'/0'/2']tpubDFS4bQGgvhftuQmUK3JhTbNTfMkBQ8Q1KP63rGzjnrmkieoYyhjLyUnMk3zwUcyH3o8bddGWZo98pcRxf81vXq6mfyiHfU47Bn1dVM9T9bM

- [x] Signet Wallet:
wsh(or_d(multi(2,[8bad162e/48'/1'/0'/2']tpubDDycBkTWNrqFprvKr2mFUJYtSDYHckrhXqv6ckMZKzhTZNgYoTMHaX5oYitKdxXTiYK1BePhKScMYFqR6qR4G3MNnyD35N9xK8kZ7Bo6xxP/<0;1>/*,[20aa8ca2/48'/1'/0'/2']tpubDFS4bQGgvhftuQmUK3JhTbNTfMkBQ8Q1KP63rGzjnrmkieoYyhjLyUnMk3zwUcyH3o8bddGWZo98pcRxf81vXq6mfyiHfU47Bn1dVM9T9bM/<0;1>/*),and_v(v:thresh(1,pkh([8bad162e/48'/1'/0'/2']tpubDDycBkTWNrqFprvKr2mFUJYtSDYHckrhXqv6ckMZKzhTZNgYoTMHaX5oYitKdxXTiYK1BePhKScMYFqR6qR4G3MNnyD35N9xK8kZ7Bo6xxP/<2;3>/*),a:pkh([20aa8ca2/48'/1'/0'/2']tpubDFS4bQGgvhftuQmUK3JhTbNTfMkBQ8Q1KP63rGzjnrmkieoYyhjLyUnMk3zwUcyH3o8bddGWZo98pcRxf81vXq6mfyiHfU47Bn1dVM9T9bM/<2;3>/*)),older(65535))))#35p4jz33

- [x] Register descriptor on Jade

- [x] Fund wallet
 - Txid: 7eb5edbbc1535083b7ed5da12b28b2a4c1fdc53153300246e84136bc7e44d689

- [x] Send to external + 1 change
  - PSBT:
cHNidP8BAH4CAAAAAYnWRH68NkHoRgIwUzHF/cGksigroV3tt4NQU8G77bV+AQAAAAD9////AiChBwAAAAAAF6kU23Nd52zttKPBHOG0u1Z4jxH/2DmHWqAHAAAAAAAiACCkI6OvL8050kIDIGV2ZiZ/TMS3UguAaDNfRHsnnkoNoAAAAAAAAQDqAgAAAAABAfTWim2zwJYwFmmIx5Ft1GuQhpex2wxJL8QKj13blcUnAQAAAAD9////Ap2IHQAAAAAAFgAUgWVhCDDB7YGx5O14QPvV3VEAMytAQg8AAAAAACIAIKMrmH6jH6ewHebBUEfSxAjrZs0zTaT0F7+g6/eLyPxgAkcwRAIgM+krx6AXY9YovmTbQ80b3jkw3wxe+qHUYIkxKXhkLmsCIGzCGIwHw4qi1df1QlDcLYOHh4IJbBnu13GUymYx4otxASECwC5uNGEIZZjCBs3YCb2zGZ7lyAITam0hIvMlMSAvzOOcFgMAAQErQEIPAAAAAAAiACCjK5h+ox+nsB3mwVBH0sQI62bNM02k9Be/oOv3i8j8YAEFhlIhApSdkZs0Rr83s1srWeIPtfNKWeuIRD8XWEvh4Y3Y/lmzIQJ0DcqCX77cOWKdzZK+SJ5BayFLyrB0+1FlYAxxcrGQVFKuc2R2qRRaDOBlF8Wrl2uPzRfzF2OKn+Mkm4isa3apFORw04kaM9O6rrWi1KEkdQQaLkbriKxsk1GIA///ALJoIgYCFscTTQka6qw+UW+K5BQgQxTWsceKzusCahFyLBOqKsgcIKqMojAAAIABAACAAAAAgAIAAIACAAAAAAAAACIGAig0gpTznKqvOPFSNA7RDI3u4aGeGDVFCfdgt98OyikNHIutFi4wAACAAQAAgAAAAIACAACAAgAAAAAAAAAiBgJ0DcqCX77cOWKdzZK+SJ5BayFLyrB0+1FlYAxxcrGQVBwgqoyiMAAAgAEAAIAAAACAAgAAgAAAAAAAAAAAIgYClJ2RmzRGvzezWytZ4g+180pZ64hEPxdYS+Hhjdj+WbMci60WLjAAAIABAACAAAAAgAIAAIAAAAAAAAAAAAAAIgICoD772OuoQP18rljT8uBkXZGa2EGGCIgTARxYlkNQHc0cIKqMojAAAIABAACAAAAAgAIAAIADAAAAAQAAACICAqCjPhl8QQFsKQfu8ivqaOu3q4BUxvGpjgmiQhEvC4b/HIutFi4wAACAAQAAgAAAAIACAACAAQAAAAEAAAAiAgLOV4HNjt/E09d2fUVBhKKBQP+PRESzM0BW8GIqCyhMEBwgqoyiMAAAgAEAAIAAAACAAgAAgAEAAAABAAAAIgIDpaTPOWhG3NEO8rdk4VEf+ZtK3WEBNqEh5fc7plDbFS4ci60WLjAAAIABAACAAAAAgAIAAIADAAAAAQAAAAA=

Comment: both output are didplayed
[image]

JamieDriver, 2024-07-01:
Thanks, I can reproduce the issue.  
It looks like it isn't recognising that the descriptor fits the input being signed, and therefore doesn't have a 'spending descriptor' with which to test the outputs to see if they're going back to the spending wallet.
(Possibly due to the same xpub being used twice with different derivation paths...)  
Leave it with me, I'll look into it.

JamieDriver, 2024-07-01:
Many thanks @edouardparis and @pythcoiner - fixed locally, will be included in next fw release.
Was indeed limited to the case where the number of keys is not equal to the number of xpubs registered with the descriptor - ie. when the same signer xpub is reused in the policy with different paths.

[image]

Cheers.

pythcoiner, 2024-07-18:
nice, will test it tmr!

pythcoiner, 2024-07-19:
It looks all good on `1.0.31-beta` ! thanks!

closed 2024-07-19 by pythcoiner

JamieDriver, 2024-07-19:
fyi: Expecting to release `1.0.31` properly in the next week or two, assuming no 'big' issues found in beta cycle and no extra functionality flagged to be merged.