Blockstream/Jade #144: Add support for M5Stick Plus 2

Sandmann21s, 2024-06-26:
I would like to request that the Jade project includes support for the M5Stick Plus 2 device. Currently, the project supports the M5Stick Plus, but due to hardware differences, some configurations and adjustments are necessary to ensure compatibility with the M5Stick Plus 2.

Motivation:
The M5Stick Plus 2 has some hardware differences compared to the M5Stick Plus, including possible changes in drivers and peripherals. Adding support for this device will expand the project's compatibility and allow more users to utilize Jade with their M5Stick Plus 2 devices.

Requested Changes:

Addition of a new board configuration for the M5Stick Plus 2 (CONFIG_BOARD_TYPE_M5_STICKC_PLUS2).
Adjustments in drivers and libraries to ensure compatibility with the M5Stick Plus 2 hardware.
Update screen and peripheral settings to reflect the specifications of the M5Stick Plus 2.
Notes:

The new board configuration should be added to the appropriate configuration files (sdkconfig.defaults, components/board/Kconfig, and components/board/board.c).
Tests should be conducted on the M5Stick Plus 2 to ensure that all main functionalities work correctly.
Feedback and suggestions are welcome.
Benefits:

Expanding the compatibility of the Jade project to include more devices.
Better support for users who own the M5Stick Plus 2.
Thank you for your consideration and I am available to assist with whatever is necessary to implement this functionality.

JamieDriver, 2024-06-27:
Duplicate of #115

Hi,
Thanks for the comprehensive request - it has been requested previously and clearly there's good arguments for supporting this device.

Please check out an outstanding PR #141 from friend of the project and contributor, @3rdIteration.  This should work on your device.  Do put any feedback you haver on that item.

We most definitely do intend to review and accept this PR into the main branch before the next fw release - unfortunately there's a bit of internal work happening in similar areas of the code, so there's a bit of a 'rebase-and-merge' delay, but we do intend to get it in when the dust settles!

Give it a go and see what you think!  :+1:

JamieDriver, 2024-08-09:
M5StackCPlus2 support to master @d48d60f3 - many thanks to @3rdIteration

closed 2024-08-09 by JamieDriver