Blockstream/Jade #147: Jade stop working with electrum

evgeniy-emelyanchyk, 2024-07-24:
Electrum version is 4.5.5
Jade hw have the latest firmware 1.0.30

I've tried to setup Jade+Electrum multisig 2/2 for BTC Testnet  and it was setup without any problem, but when I tried to sign transaction I've got the error: 

```
145.65 | E | gui.qt.main_window.[payment_megatest2] | on_error
Traceback (most recent call last):
  File "electrum/gui/qt/util.py", line 913, in run
  File "electrum/wallet.py", line 2399, in sign_transaction
  File "electrum/plugin.py", line 362, in wrapper
  File "electrum/plugin.py", line 355, in run_in_hwd_thread
  File "concurrent/futures/_base.py", line 446, in result
  File "concurrent/futures/_base.py", line 391, in __get_result
  File "concurrent/futures/thread.py", line 58, in run
  File "electrum/plugins/jade/jade.py", line 287, in sign_transaction
    multisig_name = _register_multisig_wallet(wallet, self, txout.address)
  File "electrum/plugins/jade/jade.py", line 64, in _register_multisig_wallet
    signers.append({'fingerprint': bytes.fromhex(fingerprint),
TypeError: fromhex() argument must be str, not None
347.01 | E | gui.qt.main_window.[payment_megatest2] | on_error
Traceback (most recent call last):
  File "electrum/gui/qt/util.py", line 913, in run
  File "electrum/wallet.py", line 2399, in sign_transaction
  File "electrum/plugin.py", line 362, in wrapper
  File "electrum/plugin.py", line 355, in run_in_hwd_thread
  File "concurrent/futures/_base.py", line 446, in result
  File "concurrent/futures/_base.py", line 391, in __get_result
  File "concurrent/futures/thread.py", line 58, in run
  File "electrum/plugins/jade/jade.py", line 287, in sign_transaction
    multisig_name = _register_multisig_wallet(wallet, self, txout.address)
  File "electrum/plugins/jade/jade.py", line 64, in _register_multisig_wallet
    signers.append({'fingerprint': bytes.fromhex(fingerprint),
TypeError: fromhex() argument must be str, not None
```

Then I try to setup standard wallet electrum+Jade and got another error when I try to sign trx:
```
qt.qpa.window: <QNSWindow: 0x7fe65a0b25d0; contentView=<QNSView: 0x7fe65a0b21d0; QCocoaWindow(0x6000039fb180, window=QWidgetWindow(0x6000024e1380, name="QPushButtonClassWindow"))>> has active key-value observers (KVO)! These will stop working now that the window is recreated, and will result in exceptions when the observers are removed. Break in QCocoaWindow::recreateWindowIfNeeded to debug.
1829.65 | E | gui.qt.main_window.[hw-wallet-jabe-testnet] | on_error
Traceback (most recent call last):
  File "serial/serialposix.py", line 621, in write
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "electrum/gui/qt/util.py", line 1050, in run
  File "electrum/wallet.py", line 2556, in sign_transaction
  File "electrum/plugin.py", line 406, in wrapper
  File "electrum/plugin.py", line 399, in run_in_hwd_thread
  File "concurrent/futures/_base.py", line 458, in result
  File "concurrent/futures/_base.py", line 403, in __get_result
  File "concurrent/futures/thread.py", line 58, in run
  File "electrum/plugins/jade/jade.py", line 310, in sign_transaction
    signatures = client.sign_tx(txn_bytes, jade_inputs, change)
  File "electrum/plugin.py", line 406, in wrapper
  File "electrum/plugin.py", line 396, in run_in_hwd_thread
  File "electrum/plugins/jade/jade.py", line 197, in sign_tx
    self.authenticate()
  File "electrum/plugin.py", line 406, in wrapper
  File "electrum/plugin.py", line 396, in run_in_hwd_thread
  File "electrum/plugins/jade/jade.py", line 134, in authenticate
    authenticated = self.jade.auth_user(self._network(), http_request_fn=_http_request)
  File "electrum/plugins/jade/jadepy/jade.py", line 744, in auth_user
    return self._jadeRpc('auth_user', params,
  File "electrum/plugins/jade/jadepy/jade.py", line 325, in _jadeRpc
    reply = self.jade.make_rpc_call(request, long_timeout)
  File "electrum/plugins/jade/jadepy/jade.py", line 2087, in make_rpc_call
    self.write_request(request)
  File "electrum/plugins/jade/jadepy/jade.py", line 1964, in write_request
    written += self.write(msg[written:])
  File "electrum/plugins/jade/jadepy/jade.py", line 1948, in write
    wrote = self.impl.write(bytes_)
  File "electrum/plugins/jade/jadepy/jade_serial.py", line 78, in write
    return self.ser.write(bytes_)
  File "serial/serialposix.py", line 655, in write
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured
```

JamieDriver, 2024-07-25:
I'll try to look at this when I can.
Have you isolated whether the change is due to Jade fw upgrade or due to electrum version ?

evgeniy-emelyanchyk, 2024-07-25:
Hello
unfortunately - not. I tested it a year ago and it worked perfectly fine.
Now it became a time to run in prod and it failed...

JamieDriver, 2024-09-05:
I can't reproduce - seems to work ok for me.

Aside - I have recently PR'd small changes related to signing, so you could test that branch if you're interested in that sort of thing:
See: https://github.com/spesmilo/electrum/pull/9183