Blockstream/Jade #148: Latest M5StackCPlus2 build nogo

tortxoFFoxtrot, 2024-08-12:
Hello @JamieDriver 

I am trying to restore a 24 word phrase onto the M5StackCPlus2.

The reason i had to restore is it just would not do anything in Green wallet, it said unlock and i unlocked it and it never asked for my passphrase just froze. So Ifactory reset it, set m,y custom pin server again and trying to restore it was freezing so first attempt I got to word 12 asnd it froze, then I got to word 19 and it froze.

Ok so I reflashed jade onto the unit, I successfully entered my 24 word phrase to restore, it did not ask me for a passphrase, asked me for a pin then I enter my pin and it is just stuck on "persisting pin" green wallet is just waiting for Jade to respond. totally frozen.

tortxoFFoxtrot, 2024-08-12:
@JamieDriver yea so the option to set a passphrase is completely missing. I restored my 24 word phrase using advanced setup and it never prompts me for a passphrase.

JamieDriver, 2024-08-12:
Not sure what's going on with talking to your pinserver, but from the above it sounds like Jade isn't getting a response.
In the Jade repo, you could try running `./jade_auth.py` (with your StickC-Jade plugged) and it should go through the unlock (or set-PIN) step.  IT should log the messages being passed back-and-forth between the host app and the 'jade' hw.
NOTE: this doesn't use Tor so only use this if you're happy with https.

```
jamie@athens:~/blockstream/jade$ ./jade_auth.py
Fetching jade version info over serial
Connecting...
Sending: {'method': 'get_version_info', 'id': '380985'} as cbor of size 35
Received msg: {'id': '380985', 'result': {'JADE_VERSION': '1.0.31-beta2', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'JADE_V1.1', 'JADE_FEATURES': 'DEV', 'IDF_VERSION': 'v5.2.2', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': '083AF269E820', 'BATTERY_STATUS': 5, 'JADE_STATE': 'LOCKED', 'JADE_NETWORKS': 'MAIN', 'JADE_HAS_PIN': True}}
Connected: {'JADE_VERSION': '1.0.31-beta2', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'JADE_V1.1', 'JADE_FEATURES': 'DEV', 'IDF_VERSION': 'v5.2.2', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': '083AF269E820', 'BATTERY_STATUS': 5, 'JADE_STATE': 'LOCKED', 'JADE_NETWORKS': 'MAIN', 'JADE_HAS_PIN': True}
Sending: {'method': 'auth_user', 'id': '795130', 'params': {'network': 'mainnet', 'epoch': 1723448665}} as cbor of size 63
Received msg: {'id': '795130', 'result': {'http_request': {'params': {'urls': ['https://j8d.io/get_pin', 'http://mrrxtq6tjpbnbm7vh5jt6mpjctn7ggyfy5wegvbeff3x7jrznqawlmid.onion/get_pin'], 'method': 'POST', 'accept': 'json', 'data': {'data': 'A0Z7xS9So2LniugVBl5VUqMZEVmv/bksOQBU4oBTrCN/EgEAAGTlK2KxnlRxZl//9ZrLqfnUbjwzkrjBh1fYiSl3OXMq4W7j2h7S6Cu87Tou9CGf783PUTEKh8FvbZVrukhMhRTmv6aEchcI+9RufdQ4IGzz9Z6mshS1dmlTubPOHqaKNKNsANfLJuc6QEYjYAf9LwLmwRSSC5kdQBrTtI4nG8GtKgVV8VwbBQs69eMg2/OVGSzFVGRHWW8Q/WkGynqEUmE='}}, 'on-reply': 'pin'}}}
Sending: {'method': 'pin', 'id': '674886', 'params': {'data': 'klmc0pGgRSTPlxcbuUmcdidbtDoQpC5VeFmv5Gthb9WE9uOuU8UIZMXcMnVXWj2dteJW3R6PMBEMUfKSj7opEZZeSui92GfMqaJS9lv8OUxSA0gVwyFZUH0MhJgWmiNJ'}} as cbor of size 165
Received msg: {'id': '674886', 'result': True}
Sending: {'method': 'get_xpub', 'id': '477895', 'params': {'network': 'mainnet', 'path': []}} as cbor of size 57
Received msg: {'id': '477895', 'result': 'xpub661MyMwAqRbcGJMgtWQnZ6b8Nk1YE4RkR2sAT9ZE3ovUH95wH5UxY1qkg7aRC7MdQD7YMauTncJMMHyWdDmkCeKMMoVwzJoK5DbZHHhinUQ'}
xpub661MyMwAqRbcGJMgtWQnZ6b8Nk1YE4RkR2sAT9ZE3ovUH95wH5UxY1qkg7aRC7MdQD7YMauTncJMMHyWdDmkCeKMMoVwzJoK5DbZHHhinUQ
Sending: {'method': 'sign_message', 'id': '699764', 'params': {'path': [1195487518], 'message': 'greenaddress.it      login ABCDE'}} as cbor of size 92
Received msg: {'id': '699764', 'result': 'H/bXtGN8FhkaPkLwKCZf4+AISH+SrCRnwpBNXxMxO/W5QRMDPKqJAjJGnRMqO+RusL9fQwlG7EtMOhbITiBPxWs='}
H/bXtGN8FhkaPkLwKCZf4+AISH+SrCRnwpBNXxMxO/W5QRMDPKqJAjJGnRMqO+RusL9fQwlG7EtMOhbITiBPxWs=
```

tortxoFFoxtrot, 2024-08-12:
> The option to set a passphrase at setup time has been deliberately removed - as apparently it was confusing to some people (beats me!). Once you have set-up the wallet/PIN, log out of the wallet (Session->Logout), and then you can explicitly setup the BIP39 passphrase behaviour (Options->BIP39 Passphrase).

Ah ok yep that explains it thanks

tortxoFFoxtrot, 2024-08-12:
> Not sure what's going on with talking to your pinserver, but from the above it sounds like Jade isn't getting a response. In the Jade repo, you could try running `./jade_auth.py` (with your [...]

Yea I think this is on me. My Fulcrum server was dead, and i think it actually blocked green from doing all networking, so because Green couldn't contact my fulcrum I think it ground to a halt and did not contact the pin server. After removing my defective fulcrum instance and using Blockstreams public servers all is weorking again, so i will close this, thanks for your assistance.

closed 2024-08-12 by tortxoFFoxtrot

JamieDriver, 2024-08-13:
Yes - I was going to say the address ofc depends on the bip32 path and so the wallet apps might be asking for different addresses.  But you got there before me ;-)

tortxoFFoxtrot, 2024-08-13:
> Yes - I was going to say the address ofc depends on the bip32 path and so the wallet apps might be asking for different addresses. But you got there before me ;-)

yah it threw me because I though green on both devices would act the same but then I figured hang on that's a naïve assumption I need to double check