Blockstream/Jade #149: Jade crashes when attempting to select 24 words BIP85 with index # > 1230

totallynarwhal, 2024-08-12:
On firmware 1.0.30, when I attempt to use a 24 words BIP85 index # greater than 1230, it crashes the Jade device resulting in a reboot.

Repro steps:
1. Select manual BIP39 Passphrase with Frequence Once
2. QR Pin Unlock with Blockstream's blind oracle, i.e. blkstrm.com/pn
3. Enter Passphrase to start session
4. Go to Wallet->BIP85 and choose 24 Words option
5. Use jog-wheel to increment index # towards 2000

[Expected]
I'm able to navigate and select a BIP85 index that's fairly large, i.e. 2000

[Actual]
When I navigate to ~1230, if I continue to press and hold the jog-wheel, I'll start to get garbled text in the ones and then attempting to increment the index further would then crash and reboot the device.

[image]

totallynarwhal, 2024-08-12:
On a related note - having to use the jog-wheel to manually incremental the index # 1-by-1 is also pretty tedious..

JamieDriver, 2024-08-13:
Yes, not sure what's going on there - maybe the wheel is generating events faster than they are being consumed and so eventually we run out of [some sort of] RAM or similar.  I'll try to reproduce.

Definitely agree a better way to enter large index #'s is needed.

JamieDriver, 2024-08-20:
Will investigate using the 'pin-entry' type screens... ie. where you set each digit in turn

JamieDriver, 2024-08-21:
eg:

[image]

JamieDriver, 2024-08-22:
Will switch to above for next fw release `1.0.32`

totallynarwhal, 2024-08-27:
Updated to allow index up to 999999 as well - awesome!

closed 2024-10-31 by JamieDriver