Blockstream/Jade #151: M5Stick C Plus 2 ERROR_INVALIDFW

1v9Cidjei, 2024-08-21:
Hello everyone! I'm facing a problem when trying to update my M5Stick C Plus 2. I had initially installed the @3rdIteration version, it worked perfectly, including Secure Boot. Yesterday I saw that the version had been released here in the official repository and I tried to update the firmware, but when I try to do it, the following message appears on the device and in the command line: ERROR_INVALIDFW. The signing keys are correct, I've already done tests that confirmed this. Yesterday I spent the whole day trying but without success. If anyone has any idea how to help me, I would be very grateful!

JamieDriver, 2024-08-22:
We use 'secure boot' and 'flash encryption' with the production Blockstream Jade devices, without issue.
The `sdkconfig` files we use are in the 'production/' directory.

I've not tried to use it on any sort of 'diy' hw so no idea how well it works, but can't immediately see why it shouldn't.
At what stage does that error appear ?  Before you've been asked to confirm the OTA on Jade, or after ?

1v9Cidjei, 2024-08-22:
The error appears a few seconds after I send the command 'python jade_ota.py --noagent'.

`Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\
Python 3.11.9
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd\
git version 2.44.0.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.2.2

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
    C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.11.1\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Espressif\frameworks\esp-idf-v5.2.2\tools

Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.2.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.2.2\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Espressif\frameworks\esp-idf-v5.2.2>cd jade

C:\Espressif\frameworks\esp-idf-v5.2.2\Jade>python ./tools/mkdefaults.py ./sdkconfig.defaults NDEBUG
Executing action: reconfigure
Running cmake in directory C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=./sdkconfig.defaults.tmp -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-idf-v5.2.2\Jade"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Current idf-component-manager default lock file version is 2.0.0, but found 1.0.0 in C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\dependencies.lock.esp32. Recreating lock file with the current version.
Manifest files have changed, solving dependencies.
NOTICE: Skipping optional dependency: espressif/esp_tinyusb
NOTICE: Skipping optional dependency: espressif/tinyusb
NOTICE: Skipping optional dependency: espressif/usb_host_msc
NOTICE: Skipping optional dependency: espressif/esp-dsp
.................NOTICE: Skipping optional dependency: espressif/esp_tinyusb
NOTICE: Skipping optional dependency: espressif/tinyusb
NOTICE: Skipping optional dependency: espressif/usb_host_msc
NOTICE: Skipping optional dependency: espressif/esp-dsp
[... 307 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\jade_ota.py", line 480, in <module>
    with JadeAPI.create_serial(device=args.serialport) as jade:
  File "C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\jadepy\jade.py", line 183, in __exit__
    self.disconnect(exc_type is not None)
  File "C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\jadepy\jade.py", line 274, in disconnect
    self.jade.disconnect(drain)
  File "C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\jadepy\jade.py", line 1961, in disconnect
    self.drain()
  File "C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\jadepy\jade.py", line 1975, in drain
    byte_ = self.impl.read(1)
            ^^^^^^^^^^^^^^^^^
  File "C:\Espressif\frameworks\esp-idf-v5.2.2\Jade\jadepy\jade_serial.py", line 86, in read
    return self.ser.read(n)
           ^^^^^^^^^^^^^^^^
  File "C:\Espressif\python_env\idf5.2_py3.11_env\Lib\site-packages\serial\serialwin32.py", line 295, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Acesso negado.', None, 5))

C:\Espressif\frameworks\esp-idf-v5.2.2\Jade>
[image]
`

JamieDriver, 2024-08-22:
So your device has 'secure boot' enabled, but the fw you are trying to upload is one without secure-boot compiled in, so it's (correctly) rejecting it.

At some point you must have built fw *with* secure-boot (and maybe flash-encryption?) enabled, and flashed that to your M5StickC.

From now on it will not allow you to re-flash - it will only accept fw by the 'ota' process.  The fw you try to upload must be the same config (wrt secure-boot/flash-encryption) as you had initially, and if secure-boot signed ofc.

JamieDriver, 2024-08-22:
You appear to create the build config using:
`python ./tools/mkdefaults.py ./sdkconfig.defaults NDEBUG`

I'm not sure what was in `./sdkconfig.defaults` or why you are recreating it based on itself - seems odd.
(The tool should have saved your original sdkconfig.defaults as 'sdkconfig.defaults.orig')

Also, this will not itself make a 'secure boot' config.  Did you initially make the secure-boot config with that tool ?  eg: with
`python ./tools/mkdefaults.py ./configs/sdkconfig_display_m5stickcplus2.defaults NDEBUG SECURE`  maybe ?

If so, you need to use that same 'SECURE' parameter to make a 'secure-boot' compatible binary.

The fw is probably being rejected because your running fw is 'secure' but the one you are trying to upload is not.  It can't deal with that.  Once you make it 'secure' it will only accept 'secure' fw uploads.

1v9Cidjei, 2024-08-22:
yes, I also used 'SECURE' in another command.

1v9Cidjei, 2024-08-22:
> Em algum momento você deve ter criado o fw _com_ inicialização segura (e talvez criptografia flash?) habilitada e instalado no seu M5StickC.

Yes, I activated it when I first installed it using the @3iteration repository. I also have a TT-GO T Display, I even use the same process, including signing with private keys, but in this case I was successful in updating.

1v9Cidjei, 2024-08-22:
I think it must be a difference between the repositories. As this is all kind of new to me, I didn't know if I would have problems in the future if the official repository version was released.

JamieDriver, 2024-08-23:
Ahh yes, the one you have loaded on there currently says:  `'BOARD_TYPE': 'UNKNOWN'`

This was an issue with the original PR'd branch - but this was fixed before that branch was merged, so it now would say `M5STICKCPLUS2`.

This is almost certainly the root of your problem.

You should be able to workaround this locally by commenting these two lines:
https://github.com/Blockstream/Jade/blob/f72e76ed6461a65e0010eba5b1323c7706024f53/main/process/ota_defines.h#L32-L33

Good luck!

1v9Cidjei, 2024-08-23:
> Ah, sim, o que você carregou aí atualmente diz: `'BOARD_TYPE': 'UNKNOWN'`
> 
> Este era um problema com a ramificação PR'd original - mas foi corrigido antes que a ramificação fosse mesclada, então agora diria `M5STICKCPLUS2`.
> 
> Essa é quase certamente a raiz do seu problema.
> 
> Você deve conseguir contornar isso localmente comentando estas duas linhas:
> 
> https://github.com/Blockstream/Jade/blob/f72e76ed6461a65e0010eba5b1323c7706024f53/main/process/ota_defines.h#L32-L33
> 
> Boa sorte!

Should these changes be made to the local branch that I have of the M5Stick C Plus 2? and replace the current fw with one with these new implementations, so that the original repository recognizes the board? If so, I did it and the error remains the same.

1v9Cidjei, 2024-08-23:
I apologize for not being so technical, I'm just an enthusiast trying to learn a little.

1v9Cidjei, 2024-08-23:
I think I understand, are you telling me to simply remove these two lines of code from this file? If so, I managed to do it, but in this case, every time I need to update I will have to do this? Isn't there a way to change the "BOARD_TYPE" that is already on the device? I don't know if I made myself clear...

JamieDriver, 2024-08-23:
Perfectly clear :-)

When uploading a fw Jade ensures the new fw 'BOARD_TYPE' is the same as the existing 'BOARD_TYPE'.
So yes, if we accept that your current fw thinks the board is 'UNKNOWN', you would always have to remove those two lines, every time you wanted to build a new update...

If your update worked, then we can be pretty sure this misatch was indeed the problem.

> Isn't there a way to change the "BOARD_TYPE" that is already on the device? 

Yes, but it's a little more complicated ... ;-)
So ...
1. let's *also* locally remove the check on 'same board type' ... so now (temporarily) comment/remove these lines:
https://github.com/Blockstream/Jade/blob/f72e76ed6461a65e0010eba5b1323c7706024f53/main/process/ota_util.c#L243-L246
Build that and load onto your M5StickCJade.
Now you have a fw that says it is 'UNKNOWN' but also has no check stopping us changing it ...

Then revert the changes to `ota_defines.h` (so we go back to calling it 'M5STICKCPLUS2'), *and* revert the changes just made to `ota_util.c` (ie. bringing the `BOARD_TYPE` check back).  ie. probably back to a clean `1.0.31` (or `master`, whichever you are using).  
Then build and upload that.  You should now be at a clean build, with your board reported as 'M5STICKCPLUS2', and should have no problems going forward !

:crossed_fingers:

1v9Cidjei, 2024-08-23:
> Perfeitamente claro :-)
> 
> Ao carregar um fw, Jade garante que o novo fw 'BOARD_TYPE' seja o mesmo que o 'BOARD_TYPE' existente. Então sim, se aceitarmos que seu fw atual acha que a placa é 'UNKNOWN', você sempre teria que remover essas duas linhas, toda vez que quisesse construir uma nova atualização...
> 
> Se sua atualização funcionou, então podemos ter certeza de que esse erro foi realmente o problema.
> 
> > Não há uma maneira de alterar o "BOARD_TYPE" que já está no dispositivo?
> 
> Sim, mas é um pouco mais complicado... ;-) Então...
> 
> 1. vamos _também_ remover localmente a verificação de 'mesmo tipo de placa'... então agora (temporariamente) comente/remova estas linhas:
>    https://github.com/Blockstream/Jade/blob/f72e76ed6461a65e0010eba5b1323c7706024f53/main/process/ota_util.c#L243-L246
>    
>    Construa isso e carregue no seu M5StickCJade. 
>    Agora você tem um fw que diz que é 'UNKNOWN', mas também não tem nenhuma verificação nos impedindo de alterá-lo...
> 
> Em seguida, reverta as alterações para `ota_defines.h`(então voltamos a chamá-lo de 'M5STICKCPLUS2') _e_ reverta as alterações feitas recentemente para `ota_util.c`(ou seja, trazendo a `BOARD_TYPE`verificação de volta). Ou seja, provavelmente de volta para um limpo `1.0.31`(ou `master`, o que você estiver usando). Em seguida, construa e carregue isso. Agora você deve estar em uma construção limpa, com sua placa relatada como 'M5STICKCPLUS2', e não deve ter problemas para seguir em frente! (e a verificação 'BOARD_TYPE' está de volta, que é realmente o que queremos daqui para frente [para garantir que não carreguemos uma construção para o hardware errado e potencialmente 'brick' o dispositivo!]) 🤞

Dude! Everything went perfectly! You really helped me a lot. Without your tips I practically wouldn't have managed it. Now I have a clean version that doesn't have any problems updating. Thank you very much!!!!

JamieDriver, 2024-08-23:
My pleasure!

closed 2024-08-23 by 1v9Cidjei