Blockstream/Jade #153: error updating TTGO-Tdisplay

1v9Cidjei, 2024-08-23:
Hello! When trying to update the FW of a TTGO-TDisplay I get the following error message on the device screen:
[image]

1v9Cidjei, 2024-08-23:
```
C:\Espressif\frameworks\esp-idf-v5.2.2\Jade>python jade_ota.py --noagent
No module named 'bleak'
BLE scanning/connectivity will not be available
Reading file build/jade.bin
Read 1179636 bytes
Compressing 1179636 bytes
Compressed to 788322 bytes
Deduced compressed firmware filepath: build/1.0.31-5-g561f871_ble_1179636_fw.bin
Filename suggests full firmware : 1.0.31-5-g561f871_ble_1179636_fw.bin
Got fw file of length 788322 with expected uncompressed final fw length 1179636
Final fw hash: 9dd72e8147c2e3961ab35dd4e2d28ae03089278f14c76adcf1ca142bfad4a3aa
Jade OTA over serial
Sending: {'method': 'get_version_info', 'id': '877843'} as cbor of size 35
Received msg: {'id': '877843', 'result': {'JADE_VERSION': '1.0.31-5-g561f871', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'TTGO_TDISPLAY', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v5.2.2-dirty', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': 'EC64C999BBE4', 'BATTERY_STATUS': 0, 'JADE_STATE': 'UNINIT', 'JADE_NETWORKS': 'ALL', 'JADE_HAS_PIN': False}}
Running OTA on: {'JADE_VERSION': '1.0.31-5-g561f871', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'TTGO_TDISPLAY', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v5.2.2-dirty', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': 'EC64C999BBE4', 'BATTERY_STATUS': 0, 'JADE_STATE': 'UNINIT', 'JADE_NETWORKS': 'ALL', 'JADE_HAS_PIN': False}
Sending: {'method': 'ota', 'id': '865908', 'params': {'fwsize': 1179636, 'cmpsize': 788322, 'cmphash': '165a65f88eee49182454663e5b25931c0e1c6c93cd2aaf8c47e725d47f06ebc2', 'fwhash': '9dd72e8147c2e3961ab35dd4e2d28ae03089278f14c76adcf1ca142bfad4a3aa'}} as cbor of size 138
Unhandled message received
ts Ju
Unhandled message received
  8 2016 00:22
Unhandled message received
-892800267
Unhandled message received
13
Unhandled message received
10
Unhandled message received
st:0xc (SW_CPU_RES
Unhandled message received
b'T),bo'
Unhandled message received
t:0x13 (SPI_FAS
Unhandled message received
b'_FLASH_BOOT)\r\nconfig'
Unhandled message received
ip: 0, SPIWP:0xee

Unhandled message received
lk_
Unhandled message received
rv:0
Unhandled message received
0,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00
Unhandled message received
-13
Unhandled message received
p_drv:0x00
mode:DIO, c
Unhandled message received
ock div:2
l
Unhandled message received
ad:0x3fff00c0,l
Unhandled message received
n:291
Unhandled message received
-19
Unhandled message received
13
Unhandled message received
10
Unhandled message received
oad:0x400780
Unhandled message received
-17
Unhandled message received
-17
Unhandled message received
-13
Unhandled message received
en:23824
lo
Unhandled message received
d
Unhandled message received
-813184049
Unhandled message received
-17
Unhandled message received
-49
Unhandled message received
-21
Unhandled message received
-17
Unhandled message received
-17
Unhandled message received
-13
Unhandled message received
en:4
load:0
```

1v9Cidjei, 2024-08-23:
I reset the Jade to factory settings and now the error no longer appears, but as soon as I send the command `python jade_ota.py --noagent` the device flashes the screen and appears Jade DIY and returns to the home screen without the update occurring.

1v9Cidjei, 2024-08-24:
The device is on version 1.0.31. I'm trying to update it to disable BT via firmware.

JamieDriver, 2024-09-06:
Yes, I can reproduce.
I see this in my log just before the crash:
```
Sending ota_data message 261769 as cbor of size 4133
>> I (36033) ../main/process/ota_util.c: 219: Running firmware version: 1.0.31
>> I (36033) ../main/process/ota_util.c: 231: New firmware version: 1.0.31
>> E (42143) ../main/utils/malloc_ext.h: 59: Assertion failed: ptr
>> E (42143) ../main/utils/malloc_ext.h: 59: calloc fets Jul 29 2019 12:21:46
```
Which I suggests a memory allocation error.
See also #155 

I'll look into it.

JamieDriver, 2024-09-06:
Should have a PR/branch for you to test out early next week.

JamieDriver, 2024-09-10:
If you're up for it, give [this branch](https://github.com/Blockstream/Jade/tree/reduce_mem_use) a spin.  
Hopefully it resolves the memory issues.

JamieDriver, 2024-09-12:
Merged to master.  Please close if you agree resolved.  Thanks.

closed 2024-09-18 by 1v9Cidjei