Blockstream/Jade #154: jade diy ttgo t display dont connect BT

Sandmann21s, 2024-08-24:
After a last update to version 1.0.31, the jade diy ttgo t display has serious connection errors, every time I try to insert a pin it restarts, after some tests I managed to connect by generating a seed directly through the device and connecting the cell phone only at the end of the process but it was an error that didn't happen in the past 

Before, it was so simple to connect and pair that it already worked and now you have to try several times, connecting and forgetting the device until it works without an error.

1v9Cidjei, 2024-08-24:
Strange things have been happening to me too. I tried to restore a seed and had crashes a few times, causing me to have to restart the device. Other times it doesn't even crash, it just restarts and I have to start the whole process over again. And this started after I updated to version 1.0.31. I haven't tested BT problems because I don't use it in my setup. I'm just leaving my report here because this update really seems to be a bit unstable.

JamieDriver, 2024-09-06:
I think the memory usage has gone up in the most recent fw build and that is causing issues with basic devices without additional spi/ps-ram.
Investigating.

JamieDriver, 2024-09-06:
Should have a PR/branch for you to test out early next week.

JamieDriver, 2024-09-10:
If you're up for it, give [this branch](https://github.com/Blockstream/Jade/tree/reduce_mem_use) a spin.  
Hopefully it resolves the memory issues.

JamieDriver, 2024-09-12:
Merged to master.  Please close if you agree resolved.  Thanks.

Sandmann21s, 2024-09-17:
Sorry, I was offline for a few days. I'll do some tests now and let you know if the problem starts again. The problem happened mainly when the BT was activated and I tried to set it up or connect the Jade to a device. I'll do all the tests tomorrow and let you know. Thank you in advance for your attention.

JamieDriver, 2024-09-17:
My belief is that multiple issues with tdisplay were caused by `1.0.31` using more RAM than `.30`, and ofc the tdisplay has no PSRAM so is somewhat constrained in this regard.  Also consistent with it being worse when BLE is compiled/enabled, as ofc this will have allocated RAM.
'failed to allocate memory' can happen at any point you're doing something non-trivial, so we saw multiple issues.  
I have tried to reduce the memory footprint so hoping you're issues are resolved.
Cheers.

JamieDriver, 2024-09-17:
fyi: if you don't want to have to update your esp-idf toolchain to do the test (master is updated to v5.3.1), just use [this branch](https://github.com/Blockstream/Jade/tree/reduce_mem_use) (which is master with these fixes but before the idf upgrade).

JamieDriver, 2024-10-31:
Presumed resolved

closed 2024-10-31 by JamieDriver