Blockstream/Jade #158: No hardware device detected on Linux

smartm0use, 2024-09-03:
I have flashed Jade on ESP32-S3 device but despite I have run the following command it still doesn't work:

```
sudo groupadd plugdev
sudo usermod -aG plugdev $(whoami)
sudo cp 55-usb-jade.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger
```

I take "55-usb-jade.rules" from [here](https://github.com/spesmilo/electrum/blob/master/contrib/udev/55-usb-jade.rules).

I get this message:

> No hardware device detected.
> 
> On Linux, you might have to add a new permission to your udev rules.

What could be wrong?

JamieDriver, 2024-09-03:
Those udev rules won't help  I'm afraid as they are for production jade hw (esp32) and are only looking for specific vendor-ids/product-ids associated with those units.

If you run `lsusb`, then plug (and power-on) your device, then `lsusb` again, you should see a new device has appeared.
eg, (for me, now):
`Bus 003 Device 052: ID 303a:4001` (those last are the vendor-id:product-id)

JamieDriver, 2024-09-03:
If your jade repo is on 1.0.31 or current master etc, you'll see we recently added some new ids to the serial auto-detection:
`git diff 1.0.30 ./jadepy/jade_serial.py`

```
-    JADE_DEVICE_IDS = [(0x10c4, 0xea60), (0x1a86, 0x55d4), (0x0403, 0x6001), (0x1a86, 0x7523)]
+    JADE_DEVICE_IDS = [
+            (0x10c4, 0xea60), (0x1a86, 0x55d4), (0x0403, 0x6001),
+            (0x1a86, 0x7523), (0x303a, 0x4001), (0x303a, 0x1001)]
```

JamieDriver, 2024-09-03:
In what scenario are they not detected ?

smartm0use, 2024-09-03:
I have removed rules added before and I have rerun the commands above, with different content in the rules file:

```
KERNEL=="ttyACM*", SUBSYSTEMS=="usb", ATTRS{idVendor}=="303a", ATTRS{idProduct}=="4001", MODE="0660", GROUP="plugdev", TAG+="uaccess", TAG+="udev-acl", SYMLINK+="jade%n"
```

But it still doesn't work.

I have a doubt. Currently on the screen of the device I see "Setup Jade" or "Options" buttons... and seems I can't select none of them. Pressing the two buttons changes the position of the choices but I can't select it. Pressing on the third button just restart the device. Maybe I have also misunderstand some steps to have Jade on this ESP32 device?

JamieDriver, 2024-09-03:
Yeah, not sure what board you're using, but if it's a ttgo tdisplays3 it should be a button for left, another for right, and both together as select/press.
If it's the procamera one, I think the top button is a left/right rocker, the the bottom button is select/reboot rocker - ie you must only press one half of the button to select, the other half is 'reset' - if I remember correctly anyway ...
What the buttons do depends on what hw device you've got.

In what context is it not detected ?  I mean what are you trying to detect it with ?  I agree most production applications won't be able to detect it yet - the s3 support is still wip/experimental, and most apps will need to be extended to work with it.

smartm0use, 2024-09-03:
Ok, solved the issue about selecting buttons (I'm on tdisplays3 device), thanks :)

About detecting, I'm trying to create a new wallet with hardware wallet option of Electrum.
But Electrum still can't see it...

JamieDriver, 2024-09-03:
Right - that's not totally unexpected at this time as the s3 chip uses different ids to Jade thus far and so isn't recognised - I intend to open a PR on electrum this week to address this ...

smartm0use, 2024-09-03:
Ok, look forward to see this working with Electrum too.
Thank you!

JamieDriver, 2024-09-05:
See: https://github.com/spesmilo/electrum/pull/9183

smartm0use, 2024-09-05:
Thank you!
Meanwhile, do you know if I can try to use it with another wallet?

JamieDriver, 2024-09-11:
fyi electrum have merged the PR into master.