Blockstream/Jade #159: TTGO (Lilygo) T-Display - ERRORS

angelonardone, 2024-09-03:
I'm using the config file called "sdkconfig_display_ttgo_tdisplay.defaults" as the configuration file.
I have the TTGO T-Display hardware .

when building the "firmware" an executing the `idf.py flash monitor`, in my case as `idf.py -p /dev/cu.usbserial-57060042761 flash monitor` I get the following errors:

`
 -c /Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c
In file included from /Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/syscfg/syscfg.h:9,
                 from /Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:41,
                 from /Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:23:
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: error: 'MYNEWT_VAL_BLE_L2CAP_COC_SDU_BUFF_COUNT' undeclared here (not in a function); did you mean 'MYNEWT_VAL_BLE_MESH_ADV_BUF_COUNT'?
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap_coc_priv.h:40:40: note: in expansion of macro 'MYNEWT_VAL'
   40 | #define BLE_L2CAP_SDU_BUFF_CNT        (MYNEWT_VAL(BLE_L2CAP_COC_SDU_BUFF_COUNT))
      |                                        ^~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap_coc_priv.h:43:26: note: in expansion of macro 'BLE_L2CAP_SDU_BUFF_CNT'
   43 |     struct os_mbuf *sdus[BLE_L2CAP_SDU_BUFF_CNT];
      |                          ^~~~~~~~~~~~~~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c: In function 'ble_att_svr_rx_notify':
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: error: 'MYNEWT_VAL_BLE_SM_LVL' undeclared (first use in this function); did you mean 'MYNEWT_VAL_BLE_SM_SC'?
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2945:9: note: in expansion of macro 'MYNEWT_VAL'
 2945 |     if (MYNEWT_VAL(BLE_SM_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: each undeclared identifier is reported only once for each function it appears in
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2945:9: note: in expansion of macro 'MYNEWT_VAL'
 2945 |     if (MYNEWT_VAL(BLE_SM_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c: In function 'ble_att_svr_rx_indicate':
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: error: 'MYNEWT_VAL_BLE_SM_LVL' undeclared (first use in this function); did you mean 'MYNEWT_VAL_BLE_SM_SC'?
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
/Users/angelonardone/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:3036:9: note: in expansion of macro 'MYNEWT_VAL'
 3036 |     if (MYNEWT_VAL(BLE_SM_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
[19/564] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_ead.c.obj
ninja: build stopped: subcommand failed.
`

Any Idea what could be the problem?

thanks
angelo

JamieDriver, 2024-09-06:
Looks like some problem with the esp-idf nimble/bt component...
Is your esp-idf toolchain definitely installed, the correct version, definitely up to date, etc ?

If so, the repo should build for t-display without issue.  
1. clone the repo, checkout tag (eg `1.0.31`) or master and ensure submodules up-to-date with: `git submodule update --init --recursive`
2. remove any old/stale build artifacts with: `rm -fr sdkconfig build managed_components`
3. copy the t-display config defaults: `cp configs/sdkconfig_display_ttgo_tdisplay.defaults ./sdkconfig.defaults`
4. build!   `idf.py all` (or `idf.py build flash monitor` if you want to go the whole hog)

This should work without error.

JamieDriver, 2024-10-31:
Not heard back, presumed resolved.
Please reopen with more details if still having problems building the firmware.
Many thanks.

closed 2024-10-31 by JamieDriver