Blockstream/Jade #160: Difference between fwhash and cmphash

georgantas, 2024-09-10:
Hello, I was wondering what the difference between cmphash and fwhash is: https://jadefw.blockstream.com/bin/jade1.1/index.json When I try to update my Jade from 0.1.41 to 1.0.31, my Jade seems to display the cmphash, which I believe is different from the hash that is generated following the instructions in REPRODUCIBLE.md (fwhash?). Projects like WalletScrutiny seem advertise the fwhash as well: https://walletscrutiny.com/hardware/blockstreamjade/#result

What is the difference between these two hashes? Are there any instructions on how to compute the cmphash?

Thank you.

georgantas, 2024-09-11:
Thank you.

So:

- Do some companions apps update using deltas, while others send the compressed file?

- When updating using deltas, does the Jade “dry-run” apply them to its running firmware and calculate the fwhash to show to the user?

JamieDriver, 2024-09-11:
> Do some companions apps update using deltas, while others send the compressed file?

Probably, that's up to the app.  I think the common strategy is 'prefer delta if one exists, otherwise use full fw file'.
The idea was the deltas would be less to upload and hence faster to upgrade - but in effect most version upgrades are large enough that by the time you take into account the extra time/'difficulty' of patching, the overall ota time isn't much less with a delta.  Saying that, if we did a small release (eg just one or two important bugfixes) the delta would probably be much faster to use.

NOTEs:
1. the deltas are also compressed
2. we only produce the deltas that are small enough to be worthwhile, eg. noradio <-> ble, vN-1 -> vN, maybe vN-2->vN.  On particularly large releases we don't bother producing any deltas at all (ie. when they are all slower than using the full fw image file).

> When updating using deltas, does the Jade “dry-run” apply them to its running firmware and calculate the fwhash to show to the user?

There is no "dry-run".
In all cases (full or delta):
1. the app passes the expected hashes to jade (as obtained from the index file) [*]
2. jade shows the details to the user to confirm
3. the fw is uploaded to jade, and jade verifies the hash was as initially stated (and confirmed by user) before finalising the new fw partition and flagging it as good.

4. aside: the fw signatures are also checked before the partition is flagged, and indeed they are checked every time the current fw partition is booted.

[*] - the 'cmphash' is different for every file, as it's the hash of that (compressed) file.  The 'fwhash' should be the same for every delta leading to a certain firmware -  '1.0.231 ble' say, no matter what the starting point - as that is the hash of that final fw.  That hash can't be extracted from a delta, but can be calculated from the relevant target full fw file - as explained in the previous post.

georgantas, 2024-09-11:
> the fw is uploaded to jade, and jade verifies the hash was as initially stated (and confirmed by user) before finalising the new fw partition and flagging it as good.

For this step, when only the deltas are uploaded to the Jade, does the process look something like:
1. The existing firmware on the device is copied into a new partition.
2. The deltas are applied.
3. The hash of the partition is computed and verified against the hash the user confirmed. If good, flag the partition as good. If not, abort.

JamieDriver, 2024-09-12:
More or less, although the 'copy existing', 'apply delta', and 'update hash'  happen as the delta file is being uploaded/decompressed.

ie. for full fw file:
- User confirms values (incl hash)
while bytes-to-upload:
    - upload compressed chunk of M bytes
    - update 'calculated cmphash' with M bytes
    - decompress to N bytes (expecting average N > M ofc)
    - update 'calculated fwhash' with N bytes
    - append N bytes to new fw partition
- verify final 'calculated' hashes match those initially confirmed
- check signatures
- flag partition as good for next boot
- reboot

and for delta:
- User confirms values (incl hash)
while bytes-to-upload:
    - upload compressed chunk of M bytes
    - update 'calculated cmphash' with M bytes
    - decompress to N bytes (expecting average N > M ofc)
    - copy K bytes from current partition to new partition, applying any diffs [*]
    - update 'calculated fwhash' with K bytes
    - append K bytes to new fw partition
- verify final 'calculated' hashes match those initially confirmed
- check signatures
- flag partition as good for next boot
- reboot

[*] the delta chunk of N bytes may, for example, contain info like, 'copy 1000 bytes unchanged, skip the next 100 source bytes, write these 250 bytes [\<bytes\>], copy the next 800 bytes unchanged ...'  etc.  
That's why when a delta is uploaded the progress can appear go in bursts - an instruction like 'copy the next 256kb unchanged' is a small upload but will look like a lot of new fw written, but 'then read 2kb, bytewise-add this 2kb to the original 2kb, and then write that 2kb' will be a lot slower.

NOTE: also the upload chunks are 4kb by default, but the decompression appears to run in ~12kb chunks - so we tend to see 3 or 4 'uploads' where we feed the bytes into the decompressor, but the the 'decompress' step results in no output ... then after the ~third chunk the 'decompress' step produces a ~32kb uncompressed output and a lot of writing happens.

georgantas, 2024-09-12:
Thank you for the detailed explanation, it's clear to me now :)

I hope this issue will serve as documentation for whoever has similar questions. When I get the chance, I will also make a PR to [WalletScrutiny's script](https://gitlab.com/walletscrutiny/walletScrutinyCom/-/blob/master/scripts/test/hardware/blockstreamjade.sh) to also publish the `cmphash`.

closed 2024-09-12 by georgantas

JamieDriver, 2024-09-12:
> I hope this issue will serve as documentation for whoever has similar questions. 

I'll summarise our discussion in a wiki item for that purpose!

JamieDriver, 2024-09-12:
See: [the wiki](https://github.com/Blockstream/Jade/wiki/Jade-OTA-Hashes)

georgantas, 2024-09-12:
Opened a PR on WalletScrutiny: https://gitlab.com/walletscrutiny/walletScrutinyCom/-/merge_requests/812