Blockstream/Jade #163: Unexpected hash during update

Giszmo, 2024-09-16:
I setup my jade and used Green on Android to try it out and it detected that a firmware update was available.

[image]

Sadly the file hash doesn't match what I had previously reproduced:

[image]

Is this a binary delta hash or what is it?

Our script for WalletScrutiny.com determines these hashes:

```
$ scripts/test/hardware/blockstreamjade.sh 1.0.31
...
Results:
26c918f0e7281b9cc8fdfc10baae2e5a8ac6edf0cd08e699a913daea3b97d04e  jade_1.0.31_10_ble.bin
56c3543e2929b4b450d683fcad15a3f1cc6fe0c445db4d33c102985cfcf5bc7e  jade_1.0.31_10_ble.built.bin
56c3543e2929b4b450d683fcad15a3f1cc6fe0c445db4d33c102985cfcf5bc7e  jade_1.0.31_10_ble_stripped.bin
The Jade firmware version 1.0.31 10_ble is reproducible with above hashes.
ba25801fa09ca3d0ddb1831b6384f1122f91804cd0d5c4a578aa422e4486c0da  jade_1.0.31_10_noR.bin
45c7353e9d844a549463ab65504e27fcb4a15d3877789e5d30175cefa0bd9c61  jade_1.0.31_10_noR.built.bin
45c7353e9d844a549463ab65504e27fcb4a15d3877789e5d30175cefa0bd9c61  jade_1.0.31_10_noR_stripped.bin
The Jade firmware version 1.0.31 10_noR is reproducible with above hashes.
b9c5568de7ccc4355f95fae91b35faa9ff85c95c42ef44d6151f242691a29f1a  jade_1.0.31_11_ble.bin
35da48fdb0b478a2a1f889f97dc848af0d8b2c518b79bb0b864950cd16713920  jade_1.0.31_11_ble.built.bin
35da48fdb0b478a2a1f889f97dc848af0d8b2c518b79bb0b864950cd16713920  jade_1.0.31_11_ble_stripped.bin
The Jade firmware version 1.0.31 11_ble is reproducible with above hashes.
4c88942060c01938fcdbf6be118771a9c1610a4647b98ce25e4baa813e36d437  jade_1.0.31_11_noR.bin
4065e64e34c68f55eb28e918bf7982d8f9cb4415be6fa3babbc0638b7dd57df1  jade_1.0.31_11_noR.built.bin
4065e64e34c68f55eb28e918bf7982d8f9cb4415be6fa3babbc0638b7dd57df1  jade_1.0.31_11_noR_stripped.bin
The Jade firmware version 1.0.31 11_noR is reproducible with above hashes.
```

With my device being "11" and with radio, this should be the relevant line:

```
b9c5568de7ccc4355f95fae91b35faa9ff85c95c42ef44d6151f242691a29f1a  jade_1.0.31_11_ble.bin
```

JamieDriver, 2024-09-16:
index.py:
```
  "stable": {
    "full": [
      {
        "filename": "1.0.31_noradio_987136_fw.bin",
        "version": "1.0.31",
        "config": "noradio",
        "fwsize": 987136,
        "cmphash": "cf6510018e588122def598e6be977353f07770b25e74d4cf579839500aef1eca",
        "fwhash": "4c88942060c01938fcdbf6be118771a9c1610a4647b98ce25e4baa813e36d437"
      },
      {
        "filename": "1.0.31_ble_1249280_fw.bin",
        "version": "1.0.31",
        "config": "ble",
        "fwsize": 1249280,
        "cmphash": "9803fd7cea218fcc3637f8c345003acf556d23954eeb32837bc55e8e195bbf33",
        "fwhash": "b9c5568de7ccc4355f95fae91b35faa9ff85c95c42ef44d6151f242691a29f1a"
      }
    ],
```
So teh has you show above `9803fd...3637f8...` looks like the 'cmphash' (ie the hash of the compressed [full fw] file) of "1.0.31_ble_1249280_fw.bin", which has full uncompressed hash of `b9c556...`.  
(If Jade screen says: 'Fw Hash' if it is passed [and displays] the full uncompressed fw hash, or 'File Hash:' if it receives only the compressed file hash.)  
For some reason the android app must be sending the compressed file hash and not the full fw hash (Jade accepts either to cater for older installs of various companion apps) - so initially I'd suggest it's an issue with the android app (we should have updated all our apps to always pass the full uncompressed hash).

Can I ask what version of the android app that was and I'll chase it up.
Many thanks.
J.

Giszmo, 2024-09-18:
I'm using Blockstream Green 4.0.33.

Showing the compressed file hash is kind of expected and WalletScrutiny not showing the hash was a typo in our script. Thanks for your patience. :pray:

closed 2024-09-18 by Giszmo

JamieDriver, 2024-09-19:
No worries mate.