Blockstream/Jade #164: Can't set personal blind oracle (pinserver)

petre-c, 2024-09-16:
When trying to set a personal blind oracle, [as described here](https://help.blockstream.com/hc/en-us/articles/12800132096793-Set-up-a-personal-blind-oracle), Jade doesn't display confirmation and the terminal gets stuck on Connecting..

```shell
python3 .\set_jade_pinserver.py --serialport COM7 --set-pubkey ..\blind_pin_server\server_public_key.pub --set-url http://100.65.0.5:8096
Connecting...
Sending: {'method': 'get_version_info', 'id': '406102'} as cbor of size 35
```

I successfully performed this operation several months ago, likely before firmware version 1.0.29

I have tried on two different Jades and with two different Windows 11 machines

Jade's firmware version is 1.0.31

JamieDriver, 2024-09-17:
That looks like it's related to generally connecting to Jade [from that box, at least].
Can you interact with that Jade via a) the screen menus, and b) from any other box/phone/app ?

As you can see it isn't even getting as far as trying to pass those pinserver details to Jade, it can't even connect and get the details of what is connected.

petre-c, 2024-09-18:
> That looks like it's related to generally connecting to Jade [from that box, at least].

Issue involves two different boxes, both running Windows 11.

> Can you interact with that Jade via a) the screen menus, and b) from any other box/phone/app ?

Yes - both Jades (I have tried with 2 different units) can be interacted with via both, a) [Jade's] screen menus and b) from the desktop wallet [Sparrow].

Also, one of the Jades is already set up to work with the custom pinserver (I set it up earlier) and it does work as it should - I can unlock the Jade and use it as a signing device.

petre-c, 2024-10-08:
Ended up using [Umbrel-based Personal Blind Oracle](https://apps.umbrel.com/app/blockstream-blind-oracle)

closed 2024-10-08 by petre-c

petre-c, 2024-12-02:
There is an alternative, non-BlockStream built implementation of the Jade's Pin Server -> https://github.com/Filiprogrammer/SimpleJadePinServer

Multi vendor..