Blockstream/Jade #166: Issue with ESP32-Wrover-Cam + Waveshare Pico 1.14 LCD Hat (worked with previous versions)

Drew72-ita, 2024-09-24:
Hi,

It's been a year now that I successfully used one DIY Jade, hardware
ESP32-Wrover-Cam + Waveshare Pico 1.14 LCD Hat
(see https://github.com/Blockstream/Jade/tree/master/diy/esp32-wrover-cam#connection-diagram)

Building and flashing newer version of the software always went fine, I am using the config
cp configs/sdkconfig_diycam_esp32-wrover-cam.defaults sdkconfig.defaults

The newest version(s) [I updated last time 3-4 months ago] Jade software successfully build and flash as usual but I am left with an unusable Jade due to a graphic glitch of some sort (see picture attached).

Please note that:

1) Last working Jade version 
2) The hardware was working fine (LCD, buttons, functions) just before the flashing the new built version of Jade
3) The same issue was reported to me 2-3 months ago, it was apparently solved with a slightly different ESP32-WROVER CAM Board (I remember something about the SD slot)

[image]

greenaddress, 2024-09-24:
@Drew72-ita unfortunately we don't have a ESP32-Wrover-Cam + Waveshare Pico 1.14 LCD Hat

Generally we try to support all DIY devices we have access to personally flash to and try not to break others but ultimately we can't test every combination. Maybe some other DIY contributor can help reproduce/find the problem.

Recently we changed the display library, the sdkconfig.default or Kconfig may require some tweaking.

Drew72-ita, 2024-09-24:
@greenaddress TY for the info.

FYI I rolled back to 
eps-idf 5.1.2 (482a8fb2d78e3b58eb21b26da8a5bedf90623213) 
+
jade 1.0.29
and it works again.

(since from memory this was my last working version - but I was not updating this jade since some months)

Drew72-ita, 2024-09-25:
@greenaddress 

FYI it works fine also with
eps-idf 5.1.3 (e7771c75bd1dbbfb7b3c5381be7e063b197c9734)
+
jade 1.0.30

So the issue was effectively introduced with version 1.0.31, consistent with the note that appears in jade/CHANGELOG.md:
**Change UI graphics library to esp_lcd**

greenaddress, 2024-09-26:
@Drew72-ita in your local repo you could try changing the line https://github.com/Blockstream/Jade/blob/master/main/display_hw.c#L195

try all the combinations: 
true, true; 
false, false; 
true, false; 
false, true;

once we have which combo works we should be able to fix the config.

If it's not that it could be instead be related to https://github.com/Blockstream/Jade/blob/master/main/display_hw.c#L190 or yet again https://github.com/Blockstream/Jade/blob/master/main/display_hw.c#L204 but let's try one thing at the time.

greenaddress, 2024-09-26:
maybe @3rdIteration can also provide some guidance

3rdIteration, 2024-09-26:
I have one on hand here, so I'll have a look in the next day or so. :)

3rdIteration, 2024-09-27:
I can verify that I have reproduced the issue, it is fixed by editing the config using menuconfig and editing the following from the profile in the repo.

- Invert colours
- Mirror X
- Swap X and Y

Once you do this then everything works as expected

Drew72-ita, 2024-09-27:
@greenaddress 
I successfully reproduced this fix on my hardware, thank you very much @3rdIteration ! ^__^

greenaddress, 2024-10-03:
thanks @3rdIteration for finding the problem and thanks @Drew72-ita from catching it and testing it, will look at updating the config file to reflect those changes

JamieDriver, 2024-10-03:
Cheers guys, this has been merged into master.
Please close if you agree the issue is resolved.
Many thanks.

closed 2024-10-07 by Drew72-ita