Blockstream/Jade #167: Error when trying to modify the sdkconfig file using the python ./tools/mkdefaults.py command

1v9Cidjei, 2024-10-08:
Hello, when I try to configure the sdkconfigs file using the following code: `python ./tools/mkdefaults.py ./sdkconfig.defaults NDEBUG SECURE`. at the beginning of the configuration it seems that everything will go smoothly but at the end the following error message appears:

`'rm' is not recognized as an internal or external command, operable program or batch file. Traceback (most recent call last) \PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\subprocess.py", line 413, in check_call raise CalledProcessError(retcode, cmd) subprocess.CalledProcessError: Command 'idf.py set-target esp32 && rm -f sdkconfig sdkconfig.defaults' returned non-zero exit status 1.`

I use Windows 10.

I appreciate anyone who can help me!

JamieDriver, 2024-10-08:
This script tries to remove some files - my bad I changed it from using python commands to using (linux) `rm -f` in commit 2f38d93887187626518c80d9a77679019606bb3f.
It's not related to the esp-idf upgrade - just bad timing ;-)

I will fix this now - many thanks for reporting.

In the short term you can revert that script to before I made that questionable change (presuming you are using an esp32 board):
`git checkout 2f38d938~1 ./tools/mkdefaults.py`

(If you're using an esp32s3 board you may have to wait for my fix - but I'll make it asap)

1v9Cidjei, 2024-10-08:
Thank you for your quick response and attention! I will wait for your changes and will close the topic as resolved.

Thank you very much!

JamieDriver, 2024-10-09:
Fix should be merged to master now.  Many thanks.