Blockstream/Jade #169: DIY Jade (TTGO) 1 button not responding only while BS Green is running on Win10/11

Drew72-ita, 2024-10-10:
The title is self-explanatory, see also short (50s) attached video.

The setup is DIY Jade (TTGO)  usb connected to a Windows 10/11 running last desktop application of BSGreen

The buttons works fine when BSG is not running, but one button is not working only when BSG is running.

Mac desktop BSG app is NOT affected.

See attached video.

https://github.com/user-attachments/assets/eef96770-143a-4d59-8944-c206cd6dcccc

Will test on Win11

beyondsoft978, 2024-10-13:
i have same problem, green in windows is button not work,It's normal for me to only work with Electrum

sandman21vs, 2024-11-17:
> i have same problem, green in windows is button not work,It's normal for me to only work with Electrum

you can use it on sparow wallet but on green waller the tdisplay really hasn't been working for over a year