Blockstream/Jade #170: BIP39 Passphrase Workflow is Too Confusing

wtogami, 2024-10-14:
The way Jade currently handles BIP39 passphrase is too confusing for users of BIP39 passphrases. (Firmware 1.0.31)
The user too easily loads the wrong wallet and it is not obvious how to switch to their intended passphrase protected seed.

1) Setup Jade > Advanced Setup > Restore Wallet > (seed words or QR) > Persist PIN ...
Why does **"Advanced Setup"** not ask if the user wants to use a BIP39 passphrase? They selected Advanced so why not ask? It is confusing they are forced to load the non-passphrase wallet at least once.

2) Getting to their passphrase wallet is non-intuitive. 
* Jade > Session > Logout
* Green > Overview > Logout
* Jade > Options > BIP39 Passphrase > Frequency > Always
* Green > Hardware Devices > Jade <WALLETID> > DOES NOT WORK, gets stuck, UI doesn't explain
* Green > Hardware Devices > Bitcoin on Jade <DEVID> > This gets Jade to the PIN unlock
* Jade > Enter PIN > Enter passphrase > Green shows the intended passphrase wallet

3) The concept of BIP39 Passphrase Frequency needs a rethink.
Current Frequency options:
* Never - don't use BIP39 (default)
* Once - ask for BIP39 passphrase only for the current session, it becomes "Never" in the next session
* Always - always ask for BIP39 passphrase after PIN unlock

"Once" is confusing and rarely useful. The user intends to use a passphrase or not. "Once" adds confusion because they may load their passphrase wallet once and it SEEMS TO WORK that time. It does not ask for the passphrase the next time. They might think this means the device saved the passphrase so you don't need to enter it again. But in fact the device fell back to "Never" in the next session which is never what you expect if you use a passphrase wallet.

I've witnessed first time users stumble on the meaning of "Once". On several occasions I personally forgot the meaning and was confused by loading the wrong wallet.

### What should we do about Frequency?
1. Get rid of "Once". It is confusing and not useful to anyone. This would be the simplest change alone.
2. Instead of "Once" please consider adding add "Persist" where the passphrase is saved within the device. This would be no less secure than a passphrase-less device. If we add Persist then it might help for the unlocked Options menu to make it clear you are wiping the saved passphrase if you subsequently change the BIP39 setting.
3. Consider renaming and reordering the options ...
* "Frequency" doesn't make sense. Just call it "BIP39 Passphrase" with the following sub-options.
* "Never" is fine as-is.
* "Always" might be OK as-is or maybe more clear would be "Prompt" or something?
* "Persist" becomes the third option.

alessandro-saglimbeni, 2024-10-14:
Thanks @wtogami for the feedback, very good points!

Re:
> Why does "Advanced Setup" not ask if the user wants to use a BIP39 passphrase? They selected Advanced so why not ask? It is confusing they are forced to load the non-passphrase wallet at least once.

Too many users chose Advanced Setup, set a passphrase without understanding what it was, leading to support inquiries about lost funds, because they didn't know enough. That was too prone to loss of funds. From now on if you want to use a passphrase you must be a user that knows what and why they want it, so I think we shouldn't reintroduce the passphrase in the advanced setup flow

wtogami, 2024-10-14:
> Too many users chose Advanced Setup, set a passphrase without understanding what it was

Ugh. I can see how users using it without understanding it could lead to loss of funds. The drawback is however the current UI workflow is hostile to teaching.

wtogami, 2024-10-14:
I think "Temporary" would be clearer in intent than the meaning of "Once"?
I'm an expert user and I was not alone in being confused when I thought the meaning of Once was Persist.

JamieDriver, 2024-12-17:
I hope this is improved for you in fw `1.0.33` ?
Better naming of options, and selection not resetting when you setup a new wallet.  ie.
- use options menu to set passphrase preference
- initialise/restore wallet - passphrase settings will be respected

JamieDriver, 2025-03-19:
Closing as I think we addressed your concerns ?   Please reopen if you have further suggestions - thanks.

closed 2025-03-19 by JamieDriver