Blockstream/Jade #171: No QR option on DIY Jade (TTYGO)

robolth, 2024-10-20:
I want to use a TTYGO as a barebones stateless seed signer. I followed all instructions on [https://github.com/bitcoin-tools/diyjade](https://github.com/bitcoin-tools/diyjade) and it works without any issue, except that when I reach the connect screen, there is only USB and bluetooth options, no QR option.

JamieDriver, 2024-10-21:
A standard ttgo doesn't have a camera ... ?

robolth, 2024-10-21:
Maybe I misunderstood how this works, but I don't want to scan QR code, only input seed words manually. I do want however to display QR codes to sign tx or get receiving addresses.

JamieDriver, 2024-10-21:
In some way your wallet app needs to communicate with Jade.  Jade supports USB, Bluetooth(BLE), and (if it has a camera) scanning QR codes.  QR/'airgapped' mode really only works if Jade scan QR codes (and so needs a camera).

eg., to operate sign tx's in an 'airgapped' manner:
a) Jade scans a QR code of the unsigned tx (as displayed by the wallet sw)
b) Jade displays some details (spend, fees) for the user to view/confirm
c) Jade displays a QR code of the now-signed transaction (which the phone/laptop wallet app will need to scan).

Obviously we need a camera to get the tx to sign onto Jade in the first place, Step (a).  If some app has sent the tx to Jade over the USB (or Bluetooth) connection we send the signatures back in the same way.
In your scenario, how would you intend to get the transaction details onto Jade ?

Re addresses, Jade doesn't usually generate addresses on its own - usually the wallet application on your phone or laptop would be doing that - but Jade can be used to verify these addresses.
1. the sw wallet app derives an address and displays it as a  string, and may display it as a QR code for the coin-sender to scan
2. the wallet app can send jade the address path (eg. 'm/48'/0'/'0'/2'/0/1') and Jade will display the address for the user to verify (ie to check it is the same as the sw app says it is).
3. a Jade with a camera can scan the address QR being displayed, and will check it against expected address paths and confirm it is good.

Again some information (the address path or the address itself) has to be communicated to Jade - for a Jade without a camera this is going to be via USB or BLE ...

Perhaps I'm not fully understanding your intended use case - how would you like to work with Jade, once the seed words have been entered ?

robolth, 2024-10-21:
My bad, I fundamentally misunderstood how this works, probably not enough research. Sorry for the irrelevant issue and thanks for taking the time to explain this to me in details!

closed 2024-10-21 by robolth

JamieDriver, 2024-10-22:
No problem!