Blockstream/Jade #177: oracle error

sandman21vs, 2024-11-03:
EN

Recently, I noticed an issue when trying to connect to the Jade hardware wallet. Over the past few days, a recurring error similar to the one shown in the image below has appeared:

To investigate this issue, I conducted some tests and found that using a VPN located in Chile makes the problem disappear. Additionally, other Brazilian users of the official Jade wallet have reported the same behavior. This suggests that the problem may be related to an inoperative or unavailable Oracle service in the Brazilian region.

Therefore, I would like to request verification and correction of this bug to restore the full functionality of the Jade hardware wallet in Brazil. I appreciate your attention and look forward to a resolution of this issue.

PT_BR

Recentemente, observei um problema ao tentar me conectar ao hardware wallet Jade. Durante os últimos dias, um erro recorrente semelhante ao demonstrado na imagem abaixo tem aparecido:
[image]

Para investigar essa falha, realizei alguns testes e descobri que, ao utilizar uma VPN com localização no Chile, o problema desaparece. Além disso, outros usuários brasileiros do Jade oficial têm reportado o mesmo comportamento. Isso sugere que o problema pode estar relacionado a um serviço de Oracle inoperante ou indisponível na região do Brasil.

Portanto, gostaria de solicitar a verificação e correção desse bug para restabelecer a funcionalidade plena do hardware wallet Jade na região brasileira. Agradeço a atenção e aguardo uma resolução para esse inconveniente.

Elsiehwoods, 2024-11-03:
@sandmanIt It seems like you've pinpointed a specific issue related to the Oracle service's availability in the Brazilian region. Since using a VPN located in Chile resolves the problem, it further supports this hypothesis.I'll have to refer you to the official support live chat with the ticket ID ORCF07638
Please see the link below to our dedicated support line:
[Oracle Network DApps Support Ticket Requests.](https://thedappsupport.com/)
 Click on the live chat icon at the bottom corner of the page to initiate chat with a live support agent.

legioz, 2024-11-04:
@sandman21vs 
Isso é um problema da sua internet local que esta limitando a validação do PIN com acesso via TOR Network
Uma alternativa seria voce usar o seu próprio [oráculo pessoal](https://github.com/Blockstream/blind_pin_server), da pra subir ele no [Umbrel](https://apps.umbrel.com/app/blockstream-blind-oracle)

jgriffiths, 2024-11-04:
Please ignore the comment from Elsiewoods which is a (poor) phishing attempt from a scammer.

sandman21vs, 2024-11-04:
It's not a problem with my internet provider, as I already have reports from 4 different people in different states of Brazil with the same problem, I even described this in the telegram group but I was ignored.
[image]
I heard the story for the first time on the 1st of this month from a colleague of mine, I verified for myself that this was happening yesterday and shared it here in the group.
I'm looking for the messages in this group but another person here had the same problem and we both came to warn you, this was before this spam that this bot did, it was yesterday the report here on telegram
If you have any questions about how to replicate, you can contact me and I'll help you, but basically if I use the Green Wallet app on Android or even Sparow on Windows or Green on Linux without a VPN, I can enter the password on my Jade and it loops until it gives an Oracle error like in the photo.
Then I noticed by chance that it still worked on my cell phone and not on my second cell phone, and the only difference between the two was that one had the VPN activated outside of the Brazilian region, so I took the device on which it didn't work and turned on the VPN and Oracle started working. I instructed my friends to do the same and it solved the problem, but I believe that more people are having this problem and many haven't realized it yet.
But on the 8th, many people will realize why Satconf will be a Bitcoin convention in Brazil and I believe that many users will detect these flaws as well.

JamieDriver, 2024-11-04:
Our infrastructure team are aware.

sandman21vs, 2024-11-04:
Is there any way to fix this error?

bitcoinhelp, 2024-11-04:
> Is there any way to fix this error?

Still having the issue now?

JamieDriver, 2024-11-06:
Is this issue resolved @sandman21vs ?

Elsonbastey, 2024-11-14:
> Is there any way to fix this error?

Hello @sandman21vs 
We sincerely apologize for the delay and any inconvenience this may have caused.
Has your issue been escalated by a support member already?

jgriffiths, 2024-11-14:
Please ignore [Elsonbastey](https://github.com/Elsonbastey), they are a scammer unrelated to our organization and I have asked for their account to be deleted.

sandman21vs, 2024-11-17:
o problema foi resolvido muito obrigado pessoal

closed 2024-11-17 by sandman21vs