Blockstream/Jade #179: T-Display S3 Pro don't work with USB

ulidol, 2024-11-09:
blockstream green on windows 11 does not recognize tdisplays3pro via usb

sandman21vs, 2024-11-17:
try installing the esp32 driver CP210x_Windows_Drivers 
This should resolve your Windows connection issue.

ulidol, 2024-11-17:
Thanks, I tried but it doesn't work and not even with Android via USB

JamieDriver, 2024-11-18:
tdisplays3pro is an 'esp32s3' (not an 'esp32' as usual for Jade).

We have recently added support for this chip to Jade fw, but it may be that the app version of Green does not yet recognise that chipset as a Jade.

I just checked, and it looks like the supporting version of the desktop app was released just last week, so you may need to update the Green app.  I believe you'll need version `2.0.13`.

Ley me know if this resolves the issue.

ulidol, 2024-11-18:
with tdisplay s3 everything works perfectly with both android and windows via usb, but with tdisplay s3 pro usb doesn't work, is it possible that it is the usb otg port of the tdisplay s3 pro?

ulidol, 2024-11-21:
Green version 2.0.14 resolve issue