Blockstream/Jade #181: Jade on tails

jade2024722, 2024-11-13:
When I try to use jade through tails of is not possible to communicate with the oracle.
Did anyone tried?

JamieDriver, 2024-11-13:
On your tails o/s, can you ping the host machine (`ping j8d.io` and/or `ping -4 j8d.io`)
(I assume you're using https/tls and not tor?)

Can you hit the http endpoint ?  (`wget j8d.io/get_pin` - should give `ERROR 405: METHOD NOT ALLOWED`)

jade2024722, 2024-11-13:
Here is the output:

amnesia@amnesia:~$ ping j8d.io
PING j8d.io (35.237.55.41) 56(84) bytes of data.
From 192.168.1.167 (192.168.1.167) icmp_seq=1 Destination Port Unreachable
ping: sendmsg: Operation not permitted
From 192.168.1.167 (192.168.1.167) icmp_seq=2 Destination Port Unreachable
ping: sendmsg: Operation not permitted

/usr/bin/ping cap_net_raw=epamnesia@amnesia:~$ wget j8d.io/get_pin
--2024-11-13 20:13:10--  http://j8d.io/get_pin
Resolving j8d.io (j8d.io)... 35.237.55.41
Connecting to j8d.io (j8d.io)|35.237.55.41|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://j8d.io/get_pin [following]
--2024-11-13 20:13:11--  https://j8d.io/get_pin
Connecting to j8d.io (j8d.io)|35.237.55.41|:443... connected.
HTTP request sent, awaiting response... 405 METHOD NOT ALLOWED
2024-11-13 20:13:22 ERROR 405: METHOD NOT ALLOWED.

any idea how to solve the issue?

JamieDriver, 2024-11-13:
Well, it looks like your machine can reach the oracle - what seems to be the issue ?

jade2024722, 2024-11-15:
The issue is that I can't use Jade from sparrow wallet on tails.

JamieDriver, 2024-11-18:
From the above I'd say your machine can reach the oracle.  
I assume your machine can talk to jade - I guess it must if you're getting to the point where it's prompted you to enter a pin and is trying to unlock.
Without more detailed logs its hard to know what's going on.
Can you get any sort of debug log from Sparrow ?