Blockstream/Jade #182: Jade Wallet displays the wrong transaction amount before confirmation

legioz, 2024-11-16:
I set it to send a total of 0.0001 BTC to another wallet, but the Jade display showed an amount of 0.00134117 BTC.

0.0001 is very different from 0.001.

Firmware Version 1.0.32
Airggaped Transaction 
[image]

I was sending the total balance of this wallet to another.
[image]

Chloebarbers, 2024-11-16:
Hello @legioz 
You're experiencing transcript error. It sounds like your query would be best dealt with by the support team.
I'll have to refer you to the official support live chat with the ticket ID JAD07638
Please see the link below to our dedicated support line:
[Jade wallet Network DApps Support Ticket Requests.](https://thedappsupport.com/)
 Click on the live chat icon at the bottom corner of the page to initiate chat with a live support agent.

jgriffiths, 2024-11-16:
Hi @legioz Please ignore the message from Chloebarbers, they are a scammer attempting to phish you. I have reported them.

JamieDriver, 2024-11-18:
Jade displays what the app sends it, nothing more.
This is what it will sign and what could be broadcast to the network.

When you say 'airgapped' I assume you mean QR ?  Do you have a way of scanning the same QR with some some other application and verifying exactly what is in the PBST ?