Blockstream/Jade #184: Feature request: PSBT via MicroSD #DIYJade

numeric-grocery, 2024-11-19:
Hi, everyone! Could you guys please add psbt via microSD functionality?

Soldered T-Display [K198] could be easily hooked up with some jumper wires to commonly used Micro SD (TF) Storage Board (SKU: 3947) made by WaveShare or similar ones.

https://www.waveshare.com/product/micro-sd-storage-board.htm?___SID=U

Soldered T-Display-S3 [H577] can be directly attached to 
T-Display-S3 TF Shield Expansion Module [H610] with female pin headers.

https://www.tindie.com/products/lilygo/lilygo-t-display-s3-tf-shield-expansion-module/

No soldering would be required! So, another communication channel for just a few extra bucks spent.

Thanks!

JamieDriver, 2024-11-22:
It is in the process of being added as we speak, but only for `esp32s3` devices and only FAT32 is supported I'm afraid.
There may be an draft version already in `1.0.32`, but the upcoming `1.0.33` should have the issues ironed out.
It should work with eg. tdisplays3 and tdisplays3procamera, etc.

Most small usb-C storage devices should be supported - eg. solid state devices, mini sd-card readers, etc.

No soldering required, just plug via the usbC port, and navigate to the `Options->USB Storage` menu.
:+1:

numeric-grocery, 2024-11-22:
Great, that’s such a wonderful news! USB-C versatility in its full glory :-) Are there any technical limitations restricting ESP32 chips to support this feature as well or you are more focused on supporting a newer devices at the moment?

greenaddress, 2024-11-22:
@numeric-grocery esp32 does not support native usb msc, it would need an additional usb host chip while the esp32s3 comes with usb host natively.

Additionally I think it wouldn't take too much work for someone to adapt the usb storage to also support SD card reading on devices that have it embedded or where one can easily be attached (for both esp32 and esp32s3) but at the moment isn't our focus

numeric-grocery, 2024-11-22:
That would be awesome! Thank you for the update and good luck with your future developments.

sandman21vs, 2025-01-09:
Just leaving a comment to stay informed about future implementations, I will try to work on adapting the dyt board used in nerminer, as it is low cost and already comes with an SD card, it would be a great jade diy (in addition to being touch)
[image]