Blockstream/Jade #185: Unable to connect TTGO DIY Jade via Bluetooth to green (iOS)

1-21gigasats, 2024-11-25:
Already opened this issue on green_ios. Not sure if its a problem with the DIY-Jade or with the iOS Wallet
https://github.com/Blockstream/green_ios/issues/69

 - Tried to connect 4 different DIY Jades (TTGO) with firmware 1.0.30 and 1.0.32
 - Used 2 different iOS Devices (iPhone and iPad)
 - Tried to use initialized device, to create a new seed phase and to recover
 - Tried to remove the Pairing from both, iOS Device and DIY-Jades

All tries ended in the error: "Something went wrong when pairing Jade. Remove your Jade from iOS bluetooth settings and try again"

This seems to start first after upgrading iOS from 17 to 18.1. The same devices worked last month while i was on iOS 17.

Using the devices via USB with electrum or sparrow works fine.

19201003080114, 2024-12-02:
I can confirm I have the same issue. 
"Something went wrong when pairing Jade" during BT pairing while confirming code.
- Lilygo ESP32 TDisplay S3
- IOS 18.1.1
- iPhone 14 Pro Max

GuiHiago, 2024-12-05:
Same error here,

- Lilygo Display Picoc3
- IOs 18
- Iphone 12 Mini

Torvante, 2024-12-10:
I am having the same isue on iPhone.
Any update?

LopesRs, 2024-12-10:
Same error here 
• iPhone 14 Pro 
• iOS 18.1.1
• Jade DIY 
• Version 1.0.32-24-g5

mngsrs, 2024-12-11:
Same error here
• iPhone 14 Pro
• iOS 18.1.1 **+ 18.2**
• Jade DIY / 1.031-28-g37ba7 / https://blockstream.github.io/jadediyflasher/

GuiHiago, 2024-12-11:
Guys, I couldn't find a way to configure this wallet even with green wallet PC version.

used Crowd wallet and it is working fine!

Not a resolution for Green wallet or bluethoot case, but a solution for cold wallet use!

Regards

sandman21vs, 2024-12-12:
Be careful with what @GuiHiago said, I've never seen this wallet he mentioned and I don't know if it has real support for jade or if it's a scam

jgriffiths, 2024-12-12:
Thanks @sandman21vs

Phishers and scammers are targeting Jade users via github, and they do not provide many tools to mitigate the damage besides us removing links from comments (which we've automated) and manually reporting suspicious comments when we see them. We appreciate you warning other users, thanks!

I've hidden this comment because its possible that its not malicious, but of course we don't recommend using unapproved software with Jade, and we note that support beyond communicating on issues here is only available through blockstream.com. Also a reminder that we will never ask you for your mnemonic or any other personal information, anyone who does so is trying to scam you.

GuiHiago, 2024-12-13:
Guys, my bad!

I mean Sparrow wallet ( I don't even know a wallet called Crown)

I'm deleting my won comment to keep this thread OK and going health.

I used Sparrow wallet to start my DIY Jade wallet and it is working fine.

Regards

mngsrs, 2024-12-15:
> Same error here • iPhone 14 Pro • iOS 18.1.1 **+ 18.2** • Lilygo ESP32 TDisplay S3 • Jade DIY / 1.031-28-g37ba7 / https://blockstream.github.io/jadediyflasher/

New attempt:

• fw 1.0.33-6-g041b19

• iOS 18.2
• iPhone 14 Pro
• Lilygo ESP32 TDisplay S3

**same error.**

sandman21vs, 2024-12-16:
I bring you a video of this bug happening on iOS so you know how to replicate it

https://github.com/user-attachments/assets/2dc3aace-cf36-4139-8f52-c0489ef6ca2c

LopesRs, 2024-12-25:
**Error remains to this day!!**

1-21gigasats, 2024-12-25:
Maybe @JamieDriver can give us a feedback if this issue is properly addressed? I donated about 30 DIY Jades that are probably unusable right now for the average user. Would be really nice to have an ETA here, thanks.

LopesRs, 2025-01-02:
Jade Diy doesn't work with iOS 15.6.7
Just try with 7 Plus

davigodoy, 2025-01-07:
The same error with ttgo + iPhone 11 iOS 18.2.

JamieDriver, 2025-01-07:
a) is the issues only with ios/iphone ?
b) is the issue only with tdisplay*s3* - or with the esp32 tdisplay as well ?

davigodoy, 2025-01-07:
> a) is the issues only with ios/iphone ? b) is the issue only with tdisplay_s3_ - or with the esp32 tdisplay as well ?

At this moment only with iPhone iOS 18.
I have  LilyGo T-display

beyondsoft978, 2025-01-08:
> a) is the issues only with ios/iphone ? b) is the issue only with tdisplay_s3_ - or with the esp32 tdisplay as well ?

only with iphone ios   and i used lilygo TTgo

JamieDriver, 2025-01-08:
The BLE code in jade fw is unchanged between Jade proper and generic consumer ('diy') hw, and the esp32 chip ought to be the same in this regard afaik - jade should be using a standard esp32 (v3).

I'll ping this across to the ios guys to see if they have any ideas.

davigodoy, 2025-01-08:
Updated iOS to 18.2.1 and the problem persists.

1-21gigasats, 2025-01-08:
So as we know the bug persists within old iOS versions and old DIY-Jade Versions that worked previously. So it is probably a faulty update within the iOS Green App that happened between 09/2024 and 11/2024. Could it be the newly added genuinity check in the iOS App, that is maybe incompatible with DIY-Jades?

LopesRs, 2025-01-08:
I have already opened a complaint there in the AppStore in the Green Wallet app and commented on the fact. The class here could do the same there, I think it will speed up the resolution!

jgriffiths, 2025-01-08:
> I think it will speed up the resolution!

It will not.

Please do not review bomb the app or call for others to to do. This will have the exact opposite effect that you intend, because our support staff will have to deal with negative reviews explaining that there is no official support for DIY hardware in the Green apps. If there are too many negative reviews for features we devs may be forced to stop providing DIY support.

We provide DIY support on a best effort basis because we are hackers ourselves and want to provide low cost options for everyone to store their Bitcoin more securely. If you can help with debugging or provide information that will help us resolve the issue, we welcome your contribution. Otherwise please think carefully about whether your actions are helping.

LopesRs, 2025-01-08:
Ok Sorry !!

sandman21vs, 2025-01-09:
> a) is the issues only with ios/iphone ? b) is the issue only with tdisplay_s3_ - or with the esp32 tdisplay as well ?

I tested it using ttgo t display, ttgo t display s 3 and m5 stack c2 plus
in the 3 models I tested the error was the same, I believe that somewhere else I have already posted a video testing with an official jade and an m5 where it worked on the jade and not on the mt (the video was without content and I tried to delete the bt and remove the devices from the green app)

Lowe7777, 2025-01-10:
On Android it is working normally via bluetooth.

LopesRs, 2025-01-10:
What is the latest firmware update?

JamieDriver, 2025-01-10:
`1.0.33`

LopesRs, 2025-01-16:
Any news about the failure ??

davigodoy, 2025-01-23:
New attempt with FW 1.0.33-43

LopesRs, 2025-01-27:
Davi Godoy ? Any new results??

davigodoy, 2025-01-27:
The same error. I have 2 esp32 models, and both can't pair with bluetooth.

JamieDriver, 2025-01-27:
At the moment we are considering this an ios issue, and are expecting to make Jade firmware changes.

davigodoy, 2025-01-30:
Updated to FW 1.0.33-96 and new green wallet app. The same error.

ryandemetrius, 2025-10-18:
I'm having the same BLE connection failure with iOS 18 and wanted to provide logs and testing information.

### Hardware & Software Versions

- Device: TTGO T-Display S3 (DIY Jade)
- Firmware: 1.0.36
- iPhone: iOS 18

### Connection Behavior

The connection follows this exact sequence everytime:

**1. Initial connection establishes successfully**
  - BLE connection event fires
  - Connection parameters negotiated (interval=39, latency=0, timeout=500)

**2. Pairing completes successfully**
  - NUMCMP passkey displayed and confirmed on both devices
  - Encryption enabled (encrypted=1 authenticated=1 bonded=1)
  - Identity resolved (Event 16, 27)

**3. GATT characteristics subscribed**
  - Both attr_handle 8 and 16 subscribed (reason=1)
  - MTU updated to 517

**4. Initial data exchange begins**
  - Device sends GATT indication successfully
  - Notification acknowledged (Event 13)

**5. iPhone unsubscribes and disconnects**
  - Both characteristics unsubscribed (attr_handle 8 and 16, reason=2)
  - Immediate disconnect with reason=531 (BLE_HS_ECONTROLLER)

### Logs

```
(28283) ble.c: encryption change event; status=0
(28283) ble.c: conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=1 authenticated=1 bonded=1

(28373) ble.c: subscribe event; conn_handle=1 attr_handle=16 reason=1 prevn=0 curn=0 previ=0 curi=1

(30513) ble.c: Entering gatt_chr_event op: 1 for attr: 19
(30513) ble.c: Reading from ble device
(30513) ble.c: Reading 35 bytes
(30533) NimBLE: GATT procedure initiated: indicate; att_handle=16
(30533) ble.c: notify tx received, notifying writer

(30663) ble.c: subscribe event; conn_handle=1 attr_handle=8 reason=2 prevn=0 curn=0 previ=1 curi=0
(30663) ble.c: subscribe event; conn_handle=1 attr_handle=16 reason=2 prevn=0 curn=0 previ=1 curi=0
(30663) ble.c: disconnect; reason=531
```

I've also tested the BLE connection in the configurations below, all of which had the same result:
- iOS 18 with Jade firmware v1.0.31
- iOS 18 with Jade firmware v1.0.32
- iOS 26 with Jade firmware v1.0.36

1-21gigasats, 2026-03-14:
The issue was fixed in Green iOS https://github.com/Blockstream/green_ios/commit/6d8e76de16cb9eaa9fb8ea17f01bd32e1e97310b release 5.2.0 thanks to @CaTeIM outstanding work. For details see:

https://github.com/Blockstream/green_ios/issues/93

If you find his work useful please consider leaving him some sats for his efforts, just as I did.

danielboro, 2026-03-14:
only is iOS?
I have seen the same problem in Android. but last time I tested was at 1.36

jgriffiths, 2026-03-16:
@danielboro 1.0.39 final firmware is now released.