Blockstream/Jade #186: Timeout Error When Entering Passphrase in Green Wallet App (Mobile)

sandman21vs, 2024-11-29:
When attempting to connect to my Jade wallet using the standard method in the Green Wallet app on mobile, everything works fine. However, when trying to connect with the passphrase, a timeout error occurs.

When I enter the passphrase within 5-10 seconds, I can access the app normally.
However, if the passphrase entry time exceeds 20-30 seconds, the timeout error occurs and I am unable to access the wallet.
This issue specifically happens in the Green Wallet app on mobile, both in the "Jade" version and the "Jade Official" version. I tested both versions, and both exhibit the same behavior.

Behavior in Sparrow App (PC): I tested the same process in the Sparrow app on PC, and there the passphrase connection works as expected without the timeout error, regardless of how long it takes to enter the passphrase.

Steps to Reproduce:

Connect to Jade using the Green Wallet app on mobile.
Select the option to enter the passphrase.
Enter the passphrase.
If the passphrase is entered within 5-10 seconds, the connection is successful.
If the passphras

https://github.com/user-attachments/assets/558686ab-c2bf-442a-9193-eb17db42a288

e is entered after 20-30 seconds, a timeout error occurs.
Expected Behavior: The passphrase should be accepted regardless of the time it takes to enter it, without causing a timeout error in the Green Wallet app on mobile.

Demonstration Video:
(Attach a video or link that shows the error occurring on Green Wallet on your cell phone).

Related Behavior by Others: Several users are reporting the same issue in Jade and Green Wallet related communities and forums. It appears that the timeout error mainly affects the mobile app, while the Sparrow app on PC does not have this problem.

DINOPACON, 2024-11-29:
The moment I connect the wallet, when I open the Green Wallet app, this message "Timeout - no response received for message id: 1010" keeps appearing, sometimes when I open the wallet, sometimes before it actually opens.

hiagodogrunes, 2024-11-29:
The same problem is happening to me, it is preventing me from accessing my Jade because of this passphrase bug.

JamieDriver, 2024-11-30:
I think the issue must on the app side - ie. that the app has decided that you've had long enough and that the connection has timed out.  Jade itself has no timeout here and should let you take as long as you need to enter the PIN and any phrase.  (Note the timeout is probably the total time to enter PIN+phrase?)

In any case I can pass this on to the team in question - is the issue on android or ios (or both)  ?
(Also, do you know how the Green Desktop app behaves in this regard ?)
Thanks.

sandman21vs, 2024-11-30:
The Green Desktop app still has the same bug of blocking the Jade DIY buttons, so I haven't tested it yet, but I have an official Jade and I'm going to test it with it.

usermhz, 2024-12-03:
I'm having the same problem

JamieDriver, 2024-12-04:
I will forward the details to the mobile and desktop app teams.
Many thanks.

BitcoinAcademyClub, 2024-12-04:
Enviei agora vários e-mails para eles com o teu Tópico aqui referido.

[image]

JamieDriver, 2024-12-05:
I have raised issues against the ios app, the android app,  and the desktop app to request they check their timeouts are long enough.
To be clear, it is the app timing out - there is no change we can make to Jade (ie. in this repo) which can address this.  The timeout in question is in the mobile/desktop application.

sandman21vs, 2024-12-12:
Here I have the bug replicating itself in a ttgo t display version and in the official version, (but when we started the complaint it was impossible to access, today I managed to access but it is a bit unstable because there are days when it does not work).

The bug is happening in version 1.0.33-2-g12 of jade diy with the green wallet app on android in version 4.0.38
https://github.com/user-attachments/assets/90f920ec-12ab-4272-a699-942c1240a7e9

https://github.com/user-attachments/assets/f638eac5-bcbf-466e-b522-59d3a6bccfc9

JamieDriver, 2024-12-12:
Thanks - this is an issue with the android (and maybe also ios?) apps - and not with Jade itself.  
I have raised issues on our internal system against both these repos - I will link your video above to the existing android issue - many thanks.

sandman21vs, 2025-01-09:
Some error messages continue to appear but now it is possible to access using the passphrase, I will declare it as closed because it is already functional (but a small detail, both in the official and in the DIY some small notifications from timoutr continue to appear but they are not preventing use)

closed 2025-01-09 by sandman21vs