Blockstream/Jade #189: OTA update fails to 1.0.33

jarno83, 2024-12-08:
Hardware: M5Stack M5StickC PLUS 2
Secure boot: ENABLED
Version on device: 1.0.32-24-g5a6bff1

OTA update fails with 1.0.33 and with latest master. OTA update works with installed 1.0.32 version and also from new cloned repo of 1.0.32.

Thanks :)  

OUTPUT from terminal:

python3 jade_ota.py --noagent
Reading file build/jade.bin
Read 1183744 bytes
Compressing 1183744 bytes
Compressed to 801175 bytes
Deduced compressed firmware filepath: build/1.0.33_ble_1183744_fw.bin
Filename suggests full firmware : 1.0.33_ble_1183744_fw.bin
Got fw file of length 801175 with expected uncompressed final fw length 1183744
Final fw hash: cb984898f8e8e9ef7eb4de317f2ab95ad6fbbd6ff608bd659a52b4474c955236
Jade OTA over serial
Sending: {'method': 'get_version_info', 'id': '511323'} as cbor of size 35
Received msg: {'id': '511323', 'result': {'JADE_VERSION': '1.0.32-24-g5a6bff1', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'M5STICKCPLUS2', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v5.3.1', 'CHIP_FEATURES': 'b3000000', 'EFUSEMAC': '10061C285ABC', 'BATTERY_STATUS': 5, 'JADE_STATE': 'LOCKED', 'JADE_NETWORKS': 'MAIN', 'JADE_HAS_PIN': True}}
Running OTA on: {'JADE_VERSION': '1.0.32-24-g5a6bff1', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'M5STICKCPLUS2', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v5.3.1', 'CHIP_FEATURES': 'b3000000', 'EFUSEMAC': '10061C285ABC', 'BATTERY_STATUS': 5, 'JADE_STATE': 'LOCKED', 'JADE_NETWORKS': 'MAIN', 'JADE_HAS_PIN': True}
Sending: {'method': 'auth_user', 'id': '656176', 'params': {'network': 'mainnet', 'epoch': 1733686819}} as cbor of size 63
Received msg: {'id': '656176', 'result': {'http_request': {'params': {'urls': ['https://j8d.io/get_pin', 'http://mrrxtq6tjpbnbm7vh5jt6mpjctn7ggyfy5wegvbeff3x7jrznqawlmid.onion/get_pin'], 'method': 'POST', 'accept': 'json', 'data': {'data': 'A3vPd8ZLIUzDbuBwig2dbV5A6OU2UvDs1BM1olU5/o4iCgAAAAv2ZBvKW1tyfYNLqdbO9KaMjavpsUjm+Lj/eHnmQdL7XZUcXcLK0gjQmayteHzujIrph5O639upIJyKBXlfHtRUk03dzzCpsopsswSGfSSDMIt2K5oWyzdEBAxbjBkVtUFGkb4AipQK3KIfgrTYaFViowqTGCdG9kgM6vU4xfHzfMJ86BTY/JQb3Xdb7RC7W7jVIIvvobL+EtNPK74XUXU='}}, 'on-reply': 'pin'}}}
Sending: {'method': 'pin', 'id': '337404', 'params': {'data': 'Ro+Fb/WsQJsVq50LI3qF7GUCOS+XRgN1HidURSCvwsH3aduVA38OsfZ4R+0E907Y5Ll8mGbO5vw6odHrcTVvxXA6BU1hEyAhxJXMcKLCS9ewp0/bsrDbgz/JcTz+DQHQ'}} as cbor of size 165
Received msg: {'id': '337404', 'result': True}
Sending: {'method': 'ota', 'id': '577328', 'params': {'fwsize': 1183744, 'cmpsize': 801175, 'cmphash': 'f6caeb6b55ce1023029c78046e80d7758e0434b17d128a6649f5933e96be1f75', 'extended_replies': True, 'fwhash': 'cb984898f8e8e9ef7eb4de317f2ab95ad6fbbd6ff608bd659a52b4474c955236'}} as cbor of size 156
Received msg: {'id': '577328', 'result': True}
Sending ota_data message 366566 as cbor of size 4133
Received msg: {'id': '366566', 'result': True}
Exception causing JadeAPI context exit.
<class 'AssertionError'>

  File "/home/jarno/jade/jade_ota.py", line 499, in <module>
    has_radio, bleid = ota(jade, fwcmp, fwlen, fwhash, patchlen,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jarno/jade/jade_ota.py", line 303, in ota
    result = jade.ota_update(fwcompressed, fwlength, chunksize, fwhash,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jarno/jade/jadepy/jade.py", line 544, in ota_update
    cb(written, cmplen, result if have_extended_reply else None)
  File "/home/jarno/jade/jade_ota.py", line 283, in _log_progress
    assert (extended_reply is not None) == (extended_replies is True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Draining interface...

Traceback (most recent call last):
  File "/home/jarno/jade/jade_ota.py", line 499, in <module>
    has_radio, bleid = ota(jade, fwcmp, fwlen, fwhash, patchlen,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jarno/jade/jade_ota.py", line 303, in ota
    result = jade.ota_update(fwcompressed, fwlength, chunksize, fwhash,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jarno/jade/jadepy/jade.py", line 544, in ota_update
    cb(written, cmplen, result if have_extended_reply else None)
  File "/home/jarno/jade/jade_ota.py", line 283, in _log_progress
    assert (extended_reply is not None) == (extended_replies is True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError

JamieDriver, 2024-12-09:
Options :-):
1. Pass `--no-extended-replies` to `jade_ota.py`
(a feature added in `1.0.33` that is not supported in `1.0.32` - hence the error above.  Going forward you won't need this flag - it's only needed if your jade fw is currently less than `1.0.33`).
2. Use the `jade_ota.py` script from `1.0.32` (eg in your `1.0.33` repo, `git checkout 1.0.32 jade_ota.py`)

jarno83, 2024-12-09:
> Options :-):
>
> 1. Pass `--no-extended-replies` to `jade_ota.py`
>    (a feature added in `1.0.33` that is not supported in `1.0.32` - hence the error above.  Going forward you won't need this flag - [...]

Thank you, it worked :)

closed 2024-12-09 by jarno83