Blockstream/Jade #191: Reproducible Build script failure

xrviv, 2024-12-16:
Hello team Blockstream Jade! 

Danny here from WalletScrutiny.com. I'm currently trying to build **1.0.33**

This is our current build script over at [gitlab.](https://gitlab.com/walletscrutiny/walletScrutinyCom/-/blob/master/scripts/test/hardware/blockstreamjade.sh)

This is the result when building 1.0.33:

```
=> CACHED [esp-idf 2/2] RUN cd ~/esp/esp-idf && git checkout c8fc5f643b7a7b0d3b182d3df610844e3  0.0s
 => ERROR [esp-qemu 1/2] RUN git clone --quiet --depth 1 --branch esp-develop-9.0.0-20240606 -  73.6s
------                                                                                                
 > [esp-qemu 1/2] RUN git clone --quiet --depth 1 --branch esp-develop-9.0.0-20240606 --single-branch --recursive --shallow-submodules https://github.com/espressif/qemu.git && cd qemu && git checkout abb5ce24386972e048b401f9eca10e90b8427a20:
#11 6.170 Note: switching to 'abb5ce24386972e048b401f9eca10e90b8427a20'.
#11 6.170 
#11 6.170 You are in 'detached HEAD' state. You can look around, make experimental
#11 6.170 changes and commit them, and you can discard any commits you make in this
#11 6.170 state without impacting any branches by switching back to a branch.
#11 6.170 
#11 6.170 If you want to create a new branch to retain commits you create, you may
#11 6.170 do so (now or later) by using -c with the switch command. Example:
#11 6.170 
#11 6.170   git switch -c <new-branch-name>
#11 6.170 
#11 6.170 Or undo this operation with:
#11 6.170 
#11 6.170   git switch -
#11 6.170 
#11 6.170 Turn off this advice by setting config variable advice.detachedHead to false
#11 6.170 
#11 46.92 fatal: could not read Username for 'https://github.com': No such device or address
#11 46.92 fatal: clone of 'https://github.com/Zeex/subhook.git' into submodule path '/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook' failed
#11 46.92 Failed to clone 'UnitTestFrameworkPkg/Library/SubhookLib/subhook'. Retry scheduled
#11 47.11 fatal: could not read Username for 'https://github.com': No such device or address
#11 47.11 fatal: clone of 'https://github.com/Zeex/subhook.git' into submodule path '/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook' failed
#11 47.11 Failed to clone 'UnitTestFrameworkPkg/Library/SubhookLib/subhook' a second time, aborting
#11 47.11 fatal: Failed to recurse into submodule path 'roms/edk2'
------
executor failed running [/bin/sh -c git clone --quiet --depth 1 --branch ${ESP_QEMU_BRANCH} --single-branch --recursive --shallow-submodules https://github.com/espressif/qemu.git && cd qemu && git checkout ${ESP_QEMU_COMMIT}]: exit code: 1
```

JamieDriver, 2024-12-16:
Yes, we too noticed this has recently broken - as you identify it is a transitive upstream module so not under our control.
We have already raised an issue against the upstream (https://github.com/espressif/qemu/issues/106) who say they'll pick up a fix from further upstream ... 
In the meantime they advised not cloning recursively, as that shouldn't be necessary in any case.

Yes, we need to update our Dockerfile - we have removed the '--recursive' from the qemu clone line you show above in a branch, it just hasn't been merged to master as yet.

So our line (which seems to be working well in a CI environment) is:
```
RUN git clone --quiet --depth 1 --branch ${ESP_QEMU_BRANCH} --single-branch --shallow-submodules https://github.com/espressif/qemu.git \
&& cd qemu && git checkout ${ESP_QEMU_COMMIT}
```
I will get this change merged to master asap.
(FYI you'll see I recently made a similar change [in HWI](https://github.com/bitcoin-core/HWI/commit/21f60258e06813b98a96d3046a4364e126932c0a) for their independent build of Jade firmware.

Please note this is not a problem with the release as tagged, it is something that has happened outside of what is controlled by git commits - I imagine it is now not possible to build docker environments for a large number of previous releases using the Dockerfiles as they are in the release.  When we prepared our environment to make the `1.0.33` release it all ran perfectly ...

Are you able to tweak the Dockerfile / that line locally ?

xrviv, 2024-12-16:
Thank you for the prompt response. Will try again tomorrow.

xrviv, 2024-12-17:
Thank you! Fixed! 

Change applied to WalletScrutiny build script [here](https://gitlab.com/walletscrutiny/walletScrutinyCom/-/merge_requests/934/diffs#88a752c530c80136c0a89c860d7b9fa5649326e5)

### 1.0.33 build results
```
Results:
feb8ae2404816c9933f9bc2596329ca5976224c66b3a847c168be65797351d71  jade_1.0.33_10_ble.bin
721a47ee98eb6b91e81b0b2ef2096d99e6fbd83ff25db5f37c883588905103e3  jade_1.0.33_10_ble.built.bin
721a47ee98eb6b91e81b0b2ef2096d99e6fbd83ff25db5f37c883588905103e3  jade_1.0.33_10_ble_stripped.bin
8a8a3d571a7340ddd850db9dff832a991f1b371965100b1b3258666f56ac3b0a  jade_1.0.33_10_ble.bin.gz
The Jade firmware version 1.0.33 10_ble is reproducible with above hashes.
7ad0dbbbfe33267665df5467ce1cbbbb6c3f139562177a276dbfb91d0edaca67  jade_1.0.33_10_noR.bin
684859947657a9d2763cc1f339f3fbfaaab9629ccd45a52ca6c565f48de5b706  jade_1.0.33_10_noR.built.bin
684859947657a9d2763cc1f339f3fbfaaab9629ccd45a52ca6c565f48de5b706  jade_1.0.33_10_noR_stripped.bin
3c9745b08454848c60210c34ff07a4487be9075e467d7e520e932114970405b4  jade_1.0.33_10_noR.bin.gz
The Jade firmware version 1.0.33 10_noR is reproducible with above hashes.
d76a0cf87df6900477b1fbd1512a6d0fc4bf99a7a4c80e04313eb2c1beb4d17c  jade_1.0.33_11_ble.bin
42e3ab3fd5ac0f7c915c67ffc884f33b38ebb6bc0532cb8e94f3b7187871810f  jade_1.0.33_11_ble.built.bin
42e3ab3fd5ac0f7c915c67ffc884f33b38ebb6bc0532cb8e94f3b7187871810f  jade_1.0.33_11_ble_stripped.bin
41d0cf7ce238ccd6d682942335ec3b331e4697dcff62690f3c2f795061ea289a  jade_1.0.33_11_ble.bin.gz
The Jade firmware version 1.0.33 11_ble is reproducible with above hashes.
26230eb16f4e849b98316f4ed1fa8fd3ba4e11d9ff339d5bac4e6934e315ab03  jade_1.0.33_11_noR.bin
bbd84f8f989a705c4df0ca40657324ca983a116c3041c59d4247f7d1b9033c01  jade_1.0.33_11_noR.built.bin
bbd84f8f989a705c4df0ca40657324ca983a116c3041c59d4247f7d1b9033c01  jade_1.0.33_11_noR_stripped.bin
0c87cead4a187ee90fa3f77aec1b7632b49c12fa28bae51193ab14111b9ed936  jade_1.0.33_11_noR.bin.gz
The Jade firmware version 1.0.33 11_noR is reproducible with above hashes.
```

WS review/fix/build MR is [here](https://gitlab.com/walletscrutiny/walletScrutinyCom/-/merge_requests/934)
Closing issue.

Thanks again!

closed 2024-12-17 by xrviv

JamieDriver, 2024-12-17:
Nice - many thanks.
Our next release will have that unnecessary (and currently problematic) `--recursive` flag removed, so hopefully should be better going forward.
:+1: