Blockstream/Jade #197: "OTP failed to fetch time/counter" on airgapped Jade

mamolas, 2025-01-15:
When using Jade as OTP it fails to view stored OTPs it says "OTP failed to fetch time/counter" except when connected to Green over Bluetooth.

Is there a workaround preferably over QR to tell Jade the current timestamp so OTP can work 100% offline without relying on a  companion app?  
Running latest firmware 1.0.33

JamieDriver, 2025-01-16:
It's a fair point - how to communicate an initial 'current time' to an airgapped jade so TOTP can work.

There is a (bespoke) QR that Jade will accept to set the time - we just need something to create and display it.
Maybe this is something the jade website could do, @bitcoinhelp ?

jgriffiths, 2025-01-16:
Perhaps a UX screen prompting the user to set the time when required? A QR would need to be scanned within 30s which might not be possible for some users...

JamieDriver, 2025-01-16:
I guess I was thinking a page with a qr which updated every ~5-10s ... but yes, it may be equally easy to just have a simple time entry screen on Jade (either utc only, or with a tz offset).

mamolas, 2025-01-17:
both options would fit best, manual input for users having issues scanning fast enough or to read a dynamic QR timestamp for a better UX, Jade could run a timer since the last timestamp update so it can generate several TOPTs in the same session until logout

JamieDriver, 2025-01-17:
Oh yes, once the time is set (after power-up/boot) jade then keeps time internally - it just does not have a clock that is running when powered-off, so needs to be told at every power-on or unlock.
At the moment the companion apps tell Jade the current utc time when they initially connect, and jade keeps track of it internally from then - so yes, you'd only have to scan/enter the time once per "power-session", and you should be able to generate as many OTPs as you require all the time jade is not powered-down.

mamolas, 2025-06-30:
I tried to set time offline using ur:jade-msg and CBOR structure as BC-UR as per Jade's standard but the QR is not recognized on latest firmware 1.0.35 @bitcoinhelp @JamieDriver 
[Live timestamp QR generator](https://mamolas.github.io/jadetime/)
[Source repo](https://github.com/mamolas/jadetime)
[GPT Prompts](https://github.com/copilot/share/c00913b6-0000-8452-b911-820084da0058)