Blockstream/Jade #201: Default Export Zpub

ddustin, 2025-01-20:
## Issue:
* It is extraordinarily difficult to add Jade to an existing quorum of multisig devices.

## Description:
When adding a Jade to an existing quorum of multisig devices, all that is needed is the ability to export a zpub or xpub, and then later sign a PBT.

Both of these tasks are impossible to do with the Jade itself, as well as with the Green app tied to the Jade.

They only appear possible to do with the HWI interface.

This may be because in part, the multi-QR code functionality is broken and incompatible with any of the devices I tried it with.

## Desired Behavior
* A main menu option to export the device's zpub or xpub for the device's default account
* Display this z/xpub on the device so at minimum the user can type it out on their machine
* Show this z/xpub on the screen with *manual* qr code cycling as discussed in #200
* Less important: The Green app should be able to do this as well

rgrambergs15, 2025-01-20:
@ddustin you need to load a seed to Jade using QR PIN Unlock or SeedQR - then you can go to options and display the xpub/zpub. Plenty of guides on Blockstream.com walking through all QR flows! Jade is fully supported with Sparrow with all QR workflows

arno01, 2025-01-23:
Blockstream Green (Android App) v4.1.3 does not have the option to export xPub/zPub/yPub/Output Descriptors.

Blockstream Green (Windows App) v2.0.17 -> Select (...) in the top-left corner, next to your Jade, Settings -> Watch-only -> there are your Singlesig Extended Public Keys (zpub) and Singlesig Output Descriptors.