Blockstream/Jade #206: Automatic Detect Change Addresses

besoeasy, 2025-01-27:
Hey while scanning by default it scans for 500 address, it only scans receiving address.

While we can verify change address by edit root option, my suggestion is that it should be automatically checked as well.

JamieDriver, 2025-01-27:
Jade scans in blocks of 500 the usual/standard receive address space.  
Yes, you can tweak that a bit - ie. you could add the /1/ if you wanted to scan for a change address.  
You're suggesting scanning /0/* (receive) and /1/* by default ?
ofc this is not difficult, it just makes scanning take twice as long to get to eg. index 500 (as you'd have to check 1000 addresses rather than 500).  
Perfectly doable though.

besoeasy, 2025-01-27:
@JamieDriver this is what i said on my message.

in next update we should make it automatic

hardly any overhead on code