Blockstream/Jade #207: Ability to display receive address in Jade

besoeasy, 2025-01-27:
A option to show receive address in jade itself.

i understand that it may give us already used address since jade has no way of knowing if the address has balance or if its been used.

But still it could be helpful optional feature for 90% of users.

JamieDriver, 2025-01-27:
I'm not clear when you mean.
Jade can generate and return receive addresses - the connected app in question passes the bip32 path and Jade generates the address, shows it on-screen, and if the user confirms returns it to the calling app.  
Jade really needs the wallet app to pass the bip32 path - if jade was to generate an address without that (ie generating the path to use itself), it may use a path that the wallet app does not know about, and therefore the wallet app may never see those funds - they'd be effectively lost (as in, non-trivial to recover).

closed 2025-01-27 by JamieDriver

besoeasy, 2025-01-27:
@JamieDriver well you are correct, i never thought it like that, yes people use different wallets, different paths