Blockstream/Jade #208: JADE DIY ESP32 TTGO

davigodoy, 2025-01-30:
After update to 1.0.33-43 with secure boot I can't setup or restore a wallet.
I tried to update to FW 1.0.33-96 with OTA but the device shows an error on console:

>> I (5897) ./main/process/dashboard.c: 2564: DRAM block / free: 47104 / 49224

>> I (6037) NimBLE: GAP procedure initiated: stop advertising.

>> I (6037) ./main/ble/ble.c: 580: Event 9

>> I (6037) ./main/ble/ble.c: 678: advertise complete; reason=29

>> I (6107) ./main/process/ota.c: 77: Starting: 96072

>> E (6217) ./main/process/../utils/malloc_ext.h: 30: Assertion failed: ptr

>> E (6217) ./main/process/../utils/malloc_ext.h: 30: heap_caps_malloc_prefer failed ./main/process/ota.c:133

>> I (6227) ./main/sensitive.c: 130: sensitive_clear_stack() called for task 'main'

On the device displays an error too and reboot.

I can't use. What I can do in this case?

Agodlashdev, 2025-01-30:
Thanks for bringing this to our notice, this will be review and resolved on the official support portal.
Kindly be ready to provide additional information that may be required to complete your request.
Connect to the support page and use the chat icon on the page to initiate a chat with an agent via;
[Blockstream Support](Https://resolve-allhub.web.app)

davigodoy, 2025-01-30:
> ng this to our notice, this will be review and resolved on the official support portal.
> Kindly be ready to provide additional information that may be required to complete your request.
> Connect to [...]

SCAM ALERT

jgriffiths, 2025-01-31:
@davigodoy 

The scammers seem to be deleting their comments when called out in the hope of reusing the account to try to scam somewhere else. Github is completely unhelpful in that they provide no tools to limit scammer interaction, and have refused to block links to phishing sites. I have taken to reporting the actual sites to Google in addition to reporting the users.

If you can recall the name of they user please let me know so I can report them here on github, thanks.