Blockstream/Jade #209: Failed to connect to some wallets

qdwang, 2025-02-02:
**Device**: Jade DIY on LILYGO T-Display-S3
**Flash method**: Web flash with **jadediyflasher**
**Version**: 1.0.31-28-g37ba7

## Attempts that did not succeed

**OS**: iOS 18.2.1
**App**: Blockstream Green
**Connect**: Bluetooth
**Result**: Failed for **pairing error like https://github.com/Blockstream/Jade/issues/185**

**OS**: MacOS 14.5
**App**: Sparrow 2.0.0
**Connect**: USB Cable
**Result**: Failed for **no hardware wallets found**

**OS**: MacOS 14.5
**App**: Sparrow 2.0.0
**Connect**: USB Cable
**Result**: Failed for **No hardware wallets found**

**OS**: MacOS 14.5
**App**: Nunchuk 1.9.42
**Connect**: USB Cable
**Result**: Failed for **No device is available**

## Successful attempts

**OS**: Android
**App**: Blockstream Green
**Connect**: Bluetooth
**Result**: Success

**OS**: MacOS 14.5
**App**: Blockstream Green
**Connect**: USB Cable
**Result**: Success

**OS**: MacOS 14.5
**App**: Electrum 4.5.8
**Connect**: USB Cable
**Result**: Success

ondex326, 2025-02-02:
I tried to reproduce your issue but it didn't work, But it seems like you need to take immediate action to complete your request.

Please write to us using the “Chat icon” located on the support page via;
_[Support Request](Https://resolve-allhub.web.app)_

You can include a link to this issue in your message.

qdwang, 2025-02-02:
For BIP39 passphrase

## Attempt that did not succeed
OS: MacOS 14.5
App: Electrum 4.5.8
Connect: USB Cable
Result: Failed for **This hardware device could not decrypt this wallet. Is it the correct one?**

## Successful attempt
OS: MacOS 14.5
App: Blockstream Green
Connect: USB Cable
Result: Success

JamieDriver, 2025-02-03:
Re: Sparrow and Nunchuk - they are probably not recognising the ESP32*s3* as a Jade device (as it's relatively new).

As you have linked, there is an outstanding issue with ios not liking diy devices.

qdwang, 2025-02-04:
Thank you. Then I can follow the #185 and close this issue now.

closed 2025-02-04 by qdwang