Blockstream/Jade #210: there are no relevant input to be signed

coldmarksmen, 2025-02-03:
Essentially I’m trying to consolidate UTXO and send to my own address with Bluewallet, Sparrow and whenever I sign the transaction it says "there are no relevant input to be signed". Even when I am logged in the right account in Green iOS app and try to scan the PSBT from Bluewallet or Sparrow “Could not finalize input 0” It’s weird. I wanted to try to use coin control on the Green MacOS app but my jade couldn’t be detected on the MacOS app. I am doing all of this via bluetooth because I want to maintain air gap. Thank you. Please help me what should I do from here?

JamieDriver, 2025-02-03:
I don't know the wider context, but "There are no relevant inputs to be signed" means the PSBT has been passed to Jade (via QR code?), but there do not appear to be any inputs that need signing by this Jade.

(In PSBT signing Jade iterates over all the inputs looking for ones which need a signature with the key held by this Jade - it expects there to be at least one or it shows this error.)

coldmarksmen, 2025-02-04:
Resolved! The issue was due to my Sparrow wallet being watch-only with a master fingerprint of 00000000, causing Jade to not recognize any inputs to sign. Re-importing my xpub from Jade into Sparrow fixed it. Thanks for your response!

closed 2025-02-04 by coldmarksmen

jgriffiths, 2025-02-04:
We should probably warn that Jade may not recognise all inputs/outputs if any keypaths have an empty fingerprint.

coldmarksmen, 2025-02-04:
I also did in the beginning with a watch only in Bluewallet and it had the same thing. Im not sure why because previously it always works with Bluewallet especially in the phone we have limited information about empty fingerprints.