Blockstream/Jade #215: Failure to upgrade 1.0.29 to 1.0.33

Giszmo, 2025-02-11:
I just setup a Jade with a new mnemonic and it appeared to work using Sparrow (received and sent BTC fine) but running the update CLI command is so far an unresolved struggle.

The second block should be the most helpful. This first block only shows what led up to the error I gave up at.

```bash
$ git clone https://github.com/Blockstream/Jade.git
$ cd Jade/
$ python3 -m venv .venv
$ source .venv/bin/activate
$ python3 -m pip install -r requirements.txt #if there is a requirements.txt file
$ sudo vim /etc/udev/rules.d/99-blockstream-jade.rules
$ sudo udevadm control --reload-rules
$ sudo udevadm trigger
$ python3 ./update_jade_fw.py
Current Jade fw: 1.0.29 - ble
-
Delta patches (faster)
-
Full firmware images
1)   1.0.33 - noradio
2)   1.0.33 - ble
-
Select firmware: 2
Save local copy of downloaded firmware? [y/N]y
Got fw file of length 854192 with expected uncompressed final fw length 1249280
Upload fw to connected Jade [Y/n]n
$ sha256sum 1.0.33_ble_1249280_fw.bin
41d0cf7ce238ccd6d682942335ec3b331e4697dcff62690f3c2f795061ea289a  1.0.33_ble_1249280_fw.bin
$ cat 1.0.33_ble_1249280_fw.bin.hash 
d76a0cf87df6900477b1fbd1512a6d0fc4bf99a7a4c80e04313eb2c1beb4d17c
$ python3 ./update_jade_fw.py --fwfile 1.0.33_ble_1249280_fw.bin
  File "/home/leo/StudioProjects/Jade/./update_jade_fw.py", line 248, in <module>
    verinfo = jade.get_version_info()
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 409, in get_version_info
    return self._jadeRpc('get_version_info', params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 351, in _jadeRpc
    reply = self.jade.make_rpc_call(request, long_timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2346, in make_rpc_call
    reply = self.read_response(long_timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2304, in read_response
    return self.read_cbor_message()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2251, in read_cbor_message
    message = cbor.load(self)
              ^^^^^^^^^^^^^^^

Traceback (most recent call last):
  File "/home/leo/StudioProjects/Jade/./update_jade_fw.py", line 248, in <module>
    verinfo = jade.get_version_info()
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 409, in get_version_info
    return self._jadeRpc('get_version_info', params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 351, in _jadeRpc
    reply = self.jade.make_rpc_call(request, long_timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2346, in make_rpc_call
    reply = self.read_response(long_timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2304, in read_response
    return self.read_cbor_message()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2251, in read_cbor_message
    message = cbor.load(self)
              ^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa2 in position 1: invalid start byte
[... 843 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 367, in _jadeRpc
    return self._jadeRpc(
           ^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 351, in _jadeRpc
    reply = self.jade.make_rpc_call(request, long_timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2347, in make_rpc_call
    self.validate_reply(request, reply)
  File "/home/leo/StudioProjects/Jade/jadepy/jade.py", line 2318, in validate_reply
    reply['id'] == '00' and 'error' in reply
                            ^^^^^^^^^^^^^^^^
Draining interface...
Raw:
bytearray(b'\xa2bidf586583eerror\xa2dcode9}\x01gmessagex9Cannot process message - hardware locked or uninitializedbrsuADE_VERSIONf1.0.29rJADE_OTA_MAX_CHUNK\x19\x10\x00kJADE_CONFIGcBLEjBOARD_TYPEiJADE_V1.1mJADE_FEATURESbSBkIDF_VERSIONfv5.1.2mCHIP_FEATURESh32000000hEFUSEMACl34987AB5BEC8nB')
----
Hex dump:
a262696466353836353833656572726f72a264636f6465397d01676d657373616765783943616e6e6f742070726f63657373206d657373616765202d206861726477617265206c6f636b6564206f7220756e696e697469616c697a6564627273754144455f56455253494f4e66312e302e3239724a4144455f4f54415f4d41585f4348554e4b1910006b4a4144455f434f4e46494763424c456a424f4152445f54595045694a4144455f56312e316d4a4144455f46454154555245536253426b4944465f56455253494f4e6676352e312e326d434849505f46454154555245536833323030303030306845465553454d41436c3334393837414235424543386e42
Raw:
bytearray(b'ATTERY_STATUS\x04jJADE_STATEfLOCKEDmJADE_NETWORKSdMAINlJADE_HAS_PIN\xf5bes1.0.29rJADE_OTA_MAX_CHUNK\x19\x10\x00kJADE_CONFIGcBLEjBOARD_TYPEiJADE_V1.1mJADE_FEATURESbSBkIDF_VERSIONfv5.1.2mCHIP_FEATURESh32000000hEFUSEMACl34987AB5BEC8nBATTERY_STATUS\x04jJADE_STATEfLOCKEDmJADE_NET')
----
Hex dump:
4154544552595f535441545553046a4a4144455f5354415445664c4f434b45446d4a4144455f4e4554574f524b53644d41494e6c4a4144455f4841535f50494ef5626573312e302e3239724a4144455f4f54415f4d41585f4348554e4b1910006b4a4144455f434f4e46494763424c456a424f4152445f54595045694a4144455f56312e316d4a4144455f46454154555245536253426b4944465f56455253494f4e6676352e312e326d434849505f46454154555245536833323030303030306845465553454d41436c3334393837414235424543386e424154544552595f535441545553046a4a4144455f5354415445664c4f434b45446d4a4144455f4e4554

```

Giszmo, 2025-02-11:
I tried several more times and having to unlock the Jade during the firmware upgrade appears to worsen circumstances and the "ERROR_BADDATA" screen on the device comes at varying %% of progress but 10% was the furthest I got.

JamieDriver, 2025-02-13:
It looks like the serial communications are being and the messages corrupted.
Can you try with a different usb cable ?

Giszmo, 2025-02-15:
Different cable worked.

closed 2025-02-15 by Giszmo